B-Lang-org/bsc #399: Sparse For Large Allocation Allocation

abukharmeh, 2021-08-20:
Hi, 

I have been using BSC to compile some BSV code, and when modeling large memory structures, BSC does not emit sparse hint in generated Verilog, I wonder if it possible to make it do it, as that would make it easier for some compilers/ emulators ! 

[image]

Kind regards,
Ibrahim.

quark17, 2021-08-22:
I would guess that you have written a design that uses `mkRegFile`.  BSC has not emitted the Verilog code that you've shown, because BSC does not (yet) emit arrays.  That code is in handwritten Verilog modules like `RegFile.v` that are imported into BSC and used to implement modules in the library, like `mkRegFile`.

I would think that the design that instantiates an array should specify whether it intends to use the array sparsely.  And I think that could be specified by choosing to instantiate a new module called `mkRegFileSparse` instead of the usual `mkRegFile`.  This new module could be implemented with a new Verilog file, `RegFileSparse.v`, that differs from `RegFile.v` by the addition of a `sparse` attribute.

This is not currently in the library, but you could write the module yourself and use it.  You could copy the source in `src/Libraries/Base1/RegFile.bs` and adjust it to use a `RegFileSparse.v` file that you'd also create.  The `RegFile` library is written BH/Classic, so the importing of a Verilog implementation is more complicated than if the library were written in BSV.  If you're able to write the import in BSV, I would suggest that, but otherwise it might be easier to just copy what the existing BH/Classic library does.

abukharmeh, 2021-08-23:
That's actually true, I was thinking that it didnt look like transpiled code when I was cycling home last friday.  I will do that, thank you.

closed 2021-08-23 by abukharmeh