B-Lang-org/bsc #403: Compiling bsc in MAC-OS-10 : no such file or directory: '/libgmp.a'

rpkkundu, 2021-08-28:
When I compiled bsc repository on MAC-OS, I get the following error:
clang: error: no such file or directory: '/libgmp.a'

Here is a snapshot of the compilation log: 
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: ../build/x86_64-apple-darwin19.6.0-release/lib/libyices.a(yices_locks.o) has no symbols
ranlib ../build/x86_64-apple-darwin19.6.0-release/lib/libyices.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: ../build/x86_64-apple-darwin19.6.0-release/lib/libyices.a(yices_locks.o) has no symbols
gcc -m64 -fvisibility=hidden  -Qunused-arguments -Wall -Wredundant-decls -O3 -fomit-frame-pointer -fno-stack-protector   \
	   -o ../build/x86_64-apple-darwin19.6.0-release/bin/yices ../build/x86_64-apple-darwin19.6.0-release/obj/frontend/yices.o ../build/x86_64-apple-darwin19.6.0-release/lib/libyices.a /libgmp.a
clang: error: no such file or directory: '/libgmp.a'
make[6]: *** [../build/x86_64-apple-darwin19.6.0-release/bin/yices] Error 1
make[5]: *** [dist] Error 2
make[4]: *** [dist] Error 2
Mode:     release
Platform: x86_64-apple-darwin19.6.0
./install-sh -c -d /Users/rupakkundu/REPO/B-Lang-org/bsc/src/vendor/yices/v2.6/yices2-inst/bin
./install-sh -c -d /Users/rupakkundu/REPO/B-Lang-org/bsc/src/vendor/yices/v2.6/yices2-inst/lib
./install-sh -c -d /Users/rupakkundu/REPO/B-Lang-org/bsc/src/vendor/yices/v2.6/yices2-inst/include
/usr/bin/install -c -m 664 build/x86_64-apple-darwin19.6.0-release/dist/include/* /Users/rupakkundu/REPO/B-Lang-org/bsc/src/vendor/yices/v2.6/yices2-inst/include
install: build/x86_64-apple-darwin19.6.0-release/dist/include/*: No such file or directory
make[5]: *** [install-default] Error 71
make[4]: *** [install] Error 2
make[3]: *** [install] Error 2
make[2]: *** [install] Error 2
make[1]: *** [install] Error 2
make: *** [install-src] Error 2

quark17, 2021-09-01:
This is a failure in the build of Yices, which is its own GitHub project.

It appears that the configure script has somehow picked `/` as the location for the GMP library.  On my Mac, the location is `/usr/local/lib`, which is where Homebrew has installed the library.

Have you installed GMP on your system?

I notice that the build instructions don't list GMP as a requirement -- that's an oversight, sorry!  (Although it is mentioned in the Yices README.)  I believe that, for me, libgmp was installed by Homebrew as dependency of GHC.  If you installed GHC differently, then you may not have the GMP library in a common place.  If you don't have it, can you run `brew install gmp` to install it?

rpkkundu, 2021-09-01:
Hello ,
I could find libgmp.a in /usr/local/lib . But the compiler takes the path as / . Any way to overcome this .
Thanks

quark17, 2021-09-01:
The directory `src/vendor/yices/v2.6/yices2/` is a separate repo that has been cloned as a submodule.  In the Makefile one level up, are these commands for building:
```
        (cd $(YICES_SRC) ; \
                autoconf ; \
                ./configure --prefix=$(YICES_INST) ; \
                $(MAKE) $(BUILD_ARGS); \
                $(MAKE) install \
                )
```
It is the `autoconf` and `configure` steps that find the directory for GMP.  Something has gone wrong in those steps.  Were there any warnings during those steps?

You could try deleting all the generated files in the `yices2` submodule and starting over.

Or, you could edit the generated files, to insert the proper directory.  In the `yices2` directory, there should be a generated file called `configs/make.include.x86_64-apple-darwin19.6.0` that contains lines like this:
```
# GMP libraries
STATIC_GMP=/usr/local/lib/libgmp.a
STATIC_GMP_INCLUDE_DIR=

PIC_GMP=/usr/local/lib/libgmp.a
PIC_GMP_INCLUDE_DIR=
```
In your file, It probably has `/libgmp.a` instead.

rpkkundu, 2021-09-01:
The make.include file in configs folder looks correct 
# GMP libraries
STATIC_GMP=/usr/local/lib/libgmp.a
STATIC_GMP_INCLUDE_DIR=

PIC_GMP=/usr/local/lib/libgmp.a
PIC_GMP_INCLUDE_DIR=

When I make in vices/v2.6 folder I get the following warning :

Package gmp was not found in the pkg-config search path.
Perhaps you should add the directory containing `gmp.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gmp' found
(cd yices2 ; \
		autoconf ; \
		./configure --prefix=/Users/rupakkundu/REPO/B-Lang-org/bsc/src/vendor/yices/v2.6/yices2-inst ; \
		/Library/Developer/CommandLineTools/usr/bin/make libyices_install_name=libyices.2.dylib LIBS=/libgmp.a; \
		/Library/Developer/CommandLineTools/usr/bin/make install \
		)

It could be due to these lines in Makefile 
LIBGMPA=$(shell pkg-config --variable=libdir gmp)/libgmp.a
BUILD_ARGS=libyices_install_name=libyices.2.dylib LIBS=$(LIBGMPA)

I executed make in the this folder and it executed. I then continued building from the bcc folder and looks like the build proceeded but now I get a new error as below:

----- Normal build options -----
bsc start Wed Sep 1 15:41:52 IST 2021
./update-build-version.sh
BuildVersion.hs up-to-date
./update-build-system.sh
BuildSystem.hs up-to-date
ghc -Wtabs -optl -headerpad_max_install_names -hidir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -odir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -stubdir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -main-is Main_bsc \
		-O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -iGHC -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp'  -I../vendor/stp/include -I../vendor/yices/include -L../vendor/htcl -I/opt/local/include  --make bsc -j1 +RTS -M4G -A128m -RTS "-with-rtsopts=-H256m -K10m -i1" -rtsopts -L../vendor/stp/lib -lstp -L../vendor/yices/lib -lyices 
bsc done Wed Sep 1 15:41:55 IST 2021

bluetcl start Wed Sep 1 15:41:55 IST 2021
ghc -Wtabs -optl -headerpad_max_install_names -hidir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -odir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -stubdir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -iGHC -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp'  -I../vendor/stp/include -I../vendor/yices/include -L../vendor/htcl -I/opt/local/include  --make bluetcl -j1 +RTS -M4G -A128m -RTS -c
ghc -Wtabs -optl -headerpad_max_install_names -hidir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -odir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -stubdir /Users/rupakkundu/REPO/B-Lang-org/bsc/src/comp/../../build/comp -O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -iGHC -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp'  -I../vendor/stp/include -I../vendor/yices/include -L../vendor/htcl -I/opt/local/include  --make bluetcl -j1 +RTS -M4G -A128m -RTS -L../vendor/stp/lib -lstp -L../vendor/yices/lib -lyices  -ltcl8.5 -lhtcl  \
		-o bluetcl \
		-no-hs-main \
		-x c bluetcl_Main.hsc
Linking bluetcl ...
Undefined symbols for architecture x86_64:
  "_Tcl_MainEx", referenced from:
      _main in bluetcl_Main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
`clang' failed in phase `Linker'. (Exit code: 1)
make[2]: *** [bluetcl] Error 1
make[1]: *** [install] Error 2
make: *** [install-src] Error 2

quark17, 2021-09-01:
I think it's odd that you see this message:
```
Package gmp was not found in the pkg-config search path.
```
When I run `pkg-config --variable=libdir gmp` I get the answer `/usr/local/Cellar/gmp/6.2.1/lib`, which is what `/usr/local/lib/libgmp.*` points to.

My `pkg-config` was installed with Homebrew, which is also how GMP was installed.  Have you installed these in some other way?

This failure:
```
Undefined symbols for architecture x86_64:
"_Tcl_MainEx", referenced from:
```
indicates that the linker hasn't found the Tcl library either.  This is odd, because it is supposed to use the Tcl that comes with macOS.  If you have installed another version of Tcl, maybe that's causing a problem, but I'm unsure how.

quark17, 2023-01-10:
I suspect this was fixed by PR #411, which did address the Makefile lines quoted as being the problem (LIBGMPA).  In any case, there's been no recent update, so I'm closing it.  Feel free to re-open if there are still issues.

If it was an environment issue, that's something we need to work on defining (or re-defining) for macos anyway, as indicated in issue 504.

closed 2023-01-10 by quark17