B-Lang-org/bsc #406: check-smoke: "called Tcl_FindHashEntry on deleted table" error on CentOS 7

omasanori, 2021-09-06:
Executing `make check-smoke` after `make install-src` on CentOS 7.9.2009 fails with `called Tcl_FindHashEntry on deleted table` errors. The log is pasted below:

```
$ make check-smoke
make[1]: Entering directory '/home/ogino/bsc/examples/smoke_test'

Checking Verilog generation
bsc -no-show-timestamps -no-show-version -verilog FibOne.bsv
Verilog file created: mkFibOne.v

Checking Verilog simulation
bsc -no-show-timestamps -no-show-version -vsim iverilog -e mkFibOne -o mkFibOne.vexe mkFibOne.v
Verilog binary file created: mkFibOne.vexe
./mkFibOne.vexe  > smoke_test_verilog.out

Comparing result of Bluespec Simulation
Some simulator specific difference expected
diff mkFibOne.out.expected smoke_test_verilog.out

Bluespec installation looks OK

Checking bsc compiles
bsc -no-show-timestamps -no-show-version -sim FibOne.bsv
Elaborated module file created: mkFibOne.ba

Checking compile for Bluespec simulator
bsc -no-show-timestamps -no-show-version -sim -o mkFibOne.bexe -e mkFibOne mkFibOne.ba
Bluesim object created: mkFibOne.{h,o}
Bluesim object created: model_mkFibOne.{h,o}
Simulation shared library created: mkFibOne.bexe.so
Simulation executable created: mkFibOne.bexe
./mkFibOne.bexe > smoke_test_bluesim.out
called Tcl_FindHashEntry on deleted table
called Tcl_FindHashEntry on deleted table
/bin/sh: line 1: 99564 Aborted                 ./mkFibOne.bexe > smoke_test_bluesim.out
make[1]: *** [Makefile:94: smoke_test_bluesim] Error 134
make[1]: Leaving directory '/home/ogino/bsc/examples/smoke_test'
make: *** [Makefile:63: check-smoke] Error 2
```

The version of Tcl available officially on CentOS 7 is 8.5, which may or may not be related to the error.

quark17, 2021-09-23:
I am able to reproduce this behavior.  I built on Centos7 and see the same error, so I will investigate.  FYI, I notice that the GHC package on Centos7 is too old, so I used ghcup to install, which installed version 8.10.7.

quark17, 2021-11-24:
PR #98 added a workaround for Bluetcl crash on exit.  In `src/vendor/htcl/haskell.c`, there is this code:
```
void
htcl_finalizeTclObj(Tcl_Obj* o)
{
#if defined(__APPLE__) && defined(__MACH__)
/* Workaround for https://sourceforge.net/p/tcl/bugs/4043/ */
  if (Tcl_IsShared(o) == 1)
#endif
  Tcl_DecrRefCount(o);
}
```
This workaround also works for CentOS7, for me.

CentOS7 has Tcl 8.5, as you mention.  And so does macOS.  So perhaps the preprocessor conditional should be based on the Tcl version?

My CentOS7 has Tcl 8.5.13 and my macOS 11.6.1 has Tcl version 8.5.9.  The best I can tell from the sourceforge bug referenced in the comments, this is not an issue in Tcl 8.6 -- but would it have been fixed in one of the 8.5 patches?  (Tcl 8.5.13 was released in 2012 and the bug was resolved in 2009, so why should it still occur?)

The Makefile includes `platform.mk` which sets `TCLSH` and checks that it is not empty, so we can get the version with this:
```
echo 'puts $tcl_version' | $(TCLSH)
```
although I note that `platform.sh` uses this, which might be more robust:
```
echo 'catch { puts [info tclversion]; exit 0}; exit 1' | $(TCLSH)
```

quark17, 2021-11-24:
The Tcl exit bug is also discussed on #81.

closed 2021-11-26 by quark17