B-Lang-org/bsc #407: `cd testsuite; make` is required before `make check-suite`

omasanori, 2021-09-06:
In a clean environment, `make check-suite` just after `make install-src` fails:
```
$ make check-suite
"make" -C testsuite
make[1]: Entering directory '/home/omasanori/workspace/bsc/testsuite'
rm -f groups.list
./test_list.sh Boolify ETooManySteps List ListN ListN_old PopCount bsc_examples bsv05 bugID403 cache_coherence course_lab des_core floating_point idct pong reginit rotors_control three_way_round_robin_arbiter verilog bscpong spiless-tx-bsv spiless-tx-bsv-cocoon  > groups.list
/bin/sh: 1: ./test_list.sh: not found
/home/omasanori/workspace/bsc/testsuite/suitemake.mk:124: /home/omasanori/workspace/bsc/testsuite/all_tests.mk: No such file or directory
make[1]: *** No rule to make target '/home/omasanori/workspace/bsc/testsuite/all_tests.mk'.  Stop.
make[1]: Leaving directory '/home/omasanori/workspace/bsc/testsuite'
make: *** [Makefile:67: check-suite] Error 2
```
but `cd testsuite; make` works and `make check-suite` starts to work too after doing so.

quark17, 2021-09-22:
You're right, sorry about that.  And thank you for reporting.  The issue is this code in `testsuite/suitemake.mk`:
```
# Force all_tests.mk to be built exactly once at the top level invocation of make                                                            
# rather than running perl/find once per test subdir. This means the list is always                                                          
# up to date if we add new tests, but also isn't re-calculated unnecessarily.                                                                
ifeq (0,${MAKELEVEL})
DUMMY := $(shell perl $(CONFDIR)/scripts/sort-by-time.pl $(tool) | awk '{t=t " " $$0} END{print "ALL_TESTS :=" t}' > $(CONFDIR)/all_tests.mk)
endif
```
This runs a shell command to generate the file `all_tests.mk` which specifies which test subdirectories to run.  If you run `make` from the top-level of the `testsuite` directory, then `all_tests.mk` should list all the subdirectories.  But the testsuite also support running just the tests in a subdirectory; if you `cd` into a test subdirectory and run `make check`, then `all_tests.mk` will contain only the tests in the subdirectories.  (If you run `make check`, you'll notice that it runs the infrastructure twice: once in the current directory and once for all the subdirectories.  That second steps is what uses `all_tests.mk` I presume.)

This code has a guard around it, so that it is not re-executed by a recursive call to `make`.  When we run `make -C`, that apparently counts as one level, so the guard prevents the file from being generated.  If you have previously generated the file, then `make check-suite` will "work" but it may be using the wrong contents!  (That is, if you had previously run in a subdirectory, then `all_tests.mk` would only list that subdirectory and `make check-suite` would then only run those tests.)

I suspect that the makefiles in the testsuite could be significantly cleaned up.  They have evolved over years; some of the complexity may have been introduced for features that are no longer supported; some of it may have been to workaround issues with older versions of DejaGnu; etc.  I would love if someone could take time to figure out what it's doing and clean it up.

Until then, changing the `check-suite` target to use `cd` would be a workaround.  But we could also try one of two things: Change the guard to look for whether a certain variable (unique to BSC's testsuite) is set, and have the recursive calls to make in the testsuite set that variable (so that recursive calls from outside don't affect it); or, try removing the guard and see what happens!  Recursive make is used for the parallel targets (in `suitemake.mk`) (and I'm not sure if these are even working?) and for the clean targets in `cleanonly.mk` and `realclean.mk`).

quark17, 2021-12-11:
The `all_tests.mk` file was only being used for the parallel targets (which do seem to work).  By moving the uses of `all_tests.mk` into a separate makefile (`parallel.mk`) we only need to generate it just before calling into that other makefile, in the two parallel targets.

closed 2021-12-11 by quark17