B-Lang-org/bsc #418: Preferred Verilog generation method?

Necryotiks, 2021-09-30:
Is it better to generate Verilog for an entire DUT all at once or each submodule one at a time and build the DUT manually?

czeck, 2021-11-10:
recommend to use automated methods such as bsv, instead of error-prone manual methods.  Using submodule improve the design and reuse experience.

closed 2021-11-10 by czeck

quark17, 2021-11-10:
When you run BSC on a design, it doesn't have to generate one Verilog module for the entire DUT, if that's what you're asking?  BSC can generate a hierarchy of Verilog modules, that do not need any manual buidling: a top module that instantiates submodules etc.  You can control which BSV/BH source modules map to Verilog modules using the "synthesize" attribute, or (less common) the `-g` command-line flag.

Or are you asking about invoking BSC on multiple BSV/BH source files?  While you could call BSC separately on each file, you can call BSC once on the top file and give the `-u` flag to have BSC compile all the dependencies first.

Necryotiks, 2021-11-13:
Yeah. So I can create synthesizable Verilog sub-hierarchies with that attribute provided. Can I create them across handshaking bounds(ready and enable)?

quark17, 2021-11-13:
Most boundaries are ready/enable, so yes?

A BSV source design has one or more `module..endmodule` blocks in it.  These are the boundaries that BSC can generate Verilog modules for, by putting a `(* synthesize *)` attribute on it.  This is sometimes called "separate synthesis", because BSC will separately generate that module.

When you separately generate a module, BSC generates code for it without knowing anything about the context (or contexts) where it will appear.  So BSC cannot optimize the logic in the module with logic in the parent.  Similiarly, each generated module has its own scheduler, so BSC may have to conservatively schedule the rules -- to allow it to be used in any possible context, even though it may only be instantiated in a context that allows a freer scheduling.  So there can (at some module boundaries) be benefit to not separately synthesizing.  However, for large designs, there may be scaling issues that prevent BSC from compiling the entire design at once, and so it helps to break it into smaller pieces.

However, as hardware designer, I usually have in mind what hardware I want to generate.  So I know what Verilog modules I want, and I write my BSV design with that in mind and the choice of `synthesize` boundaries is driven by what I want to create.

The `synthesize` attribute can be placed on most modules.  There are a few limitations.  If your module has arguments that aren't parameters or simple ports -- for example, if you use high-level features, like passing in another module or a function or an interface -- then BSC will report an error that it can't synthesize that module.  Another limitation is if you have a type parameter in your module (aka it's "polymorphic") -- for example, if your methods can take any type `t`.  BSC can't synthesize that; you would need to create a specific instance for each type that you want a module for.  Right now, BSC won't synthesize for any type variables, although there are some situations that BSC could be improved to support.  For example, BSC ought to be able to generate for most modules with `Bit#(n)`.  (The exception would be if the parameter `n` is used to generate logic, say to replicate some logic `n` times.)

Does that help?

I would say again, as a hardware designer, when I'm using BSV, I have in mind the hardware that I want to generate, and I know that there will be boundaries in my design where I care about the timing of the communication across those boundaries.  For example, there will be boundaries where I want to make sure the two sides are registered.  Or I want to avoid combinational paths that go from A to B and back to A.  And so on.  These affect the interface types that I use for those modules, and it affects how I write the modules to implement their interfaces.  And typically these are are also boundaries that I will separately synthesize.  But other kinds of boundaries can be separately synthesized too.

Necryotiks, 2021-11-14:
This has been super helpful, thank you.