B-Lang-org/bsc #420: bsc crash when bs method arg name annotation doesn't match method arity

mieszko, 2021-10-02:
if the arg_names annotation in a classic interface declaration has fewer names than the method has arguments, bsc crashes without any useful information:

```
bsc -verilog BugMethodArgNamesArity.bs
Prelude.!!: index too large
```

BugMethodArgNamesArity.bs is:

```
package BugMethodArgNamesArity where

interface ArityBug =
    foo :: Bool -> Bool -> Action {-# arg_names = [bar] #-}

{-# verilog mkArityBug #-}
mkArityBug :: Module ArityBug
mkArityBug = module
    interface
        foo _ _ = _
```

this specific crash occurs when `!!` is used in the definition of `i'` inside `iExpandMethodLam` at IExpand.hs:1065, although at least the `!!` in the definition of `m_origion_type` below has the exact same problem.

iirc the IExpand state monad handles errors, so it should be possible to check in `iExpandMethodLam` and emit a more informative error with a reasonable position; failing that, a quick-hack ICE with a position would also work.

quark17, 2021-10-03:
Thanks for submitting this!  Since BSV was the only supported language for a while, the user experience there has been better tested and has better error checking, than BH/Classic.  In this case, the error can't occur in BSV because the declaration of each argument type, name, and port re-naming is done together.  See `pMethodPrototype` in `CVParser.lhs` -- where you'll note that it _does_ check that there are no duplicate names in the port re-naming list.

I would not wait until `IExpand` to give a user error for this; if the user isn't yet generating code, then the file with this interface would compile fine, and the user wouldn't get an error until later, possibly while compiling a different file.  The check should be early enough to apply, even if code-gen is not happening.  I would be in favor of adding the check to the BH/Classic parser (see `pIfcPragmas` in `CParser.hs`).  Putting it in a latter stage is a possibility -- including a new stage after the parsers, to do common checking -- but I think it's best to do it in the parser, where we have more knowledge of the particular syntax and how the error situation might have occurred.

It'd also be worth detecting duplicates in the list, in the BH/Classic parser, as the BSV parser does.  There _is_ checking later (in `GenWrap`) that will detect duplicate port names on generated modules (and this is needed for the more complicated cases of module arguments vs interface ports, or even just interface ports at different points in the type tree, possibly from composing separate interface type declarations).  But, again, I think it's good to catch the simple case in the parser.  This code:
```
foo :: Bool -> Bool -> Action {-# arg_names = [bar, bar] #-}
```
leads to this error currently:
```
Error: "Test2.bs", line 4, column 4: (G0055)
  Method `foo' generates a port with name `foo_bar' which conflicts with a
  port of the same name generated by method `foo' at location "Test2.bs", line
  4, column 4.
```
With fixes like this (checking in the parser or in an early stage), then the list index (with `!!`) won't fail.  However, changing the use of `!!` to something that reports an internal error wouldn't be a bad idea -- the same way that we have defined `headOrErr` (and related functions) in `Util.hs`.  I don't immediately see an `indexOrErr` style of function, so I don't think we have one yet.  And I see that there are a number of places where the compiler uses `!!`.

mieszko, 2021-10-04:
Agreed that earlier is better — technically it's not a parse error, but it's easy to kill it there.

I think doing that does require some rototilling b/c the parser combinators that the classic parser uses (`Libs/Parse.hs`) only support one kind of error: _found x, expected y, z, u, or w_. But it wouldn't be too hard to hack `Parse.hs` to add an alternative arm to the `FailAt` datatype with an `ErrMsg` inside, and carry it through the various functions that propagate `FailAt`.

quark17, 2021-10-04:
Ah, I'd forgotten about limitations of the parser combinators.  Instead of changing the parser library, we could add a separate step that runs on the result of the BH/Classic parser, sanity checking for things like this -- assuming that the result has all the position data that we want (which I assume there would be, on the rename strings); if there's more position available at that point inside the parser, that could argue for doing it in the parser itself.

The parser `CParser::pPackage` is run in `Depend::parseSrc`, so the check could be there.  Or `pPackage` could be pushed down a level, so that it's actually a wrapper around the combinator calls; I see that `Depend` already defines `chkParse` as a wrapper that it uses to convert the bare errors into `EMsg` (using a function that `CPackage` itself defines and doesn't use, only exports!).

mieszko, 2021-10-11:
Personally, I think it would be very convenient to be able to check some stuff and report a broader range of errors from CParse. In theory it's just a parser and one ought to do the remaining checks on the CSyntax tree, but it's sometimes just so much more convenient to do it while parsing :)