B-Lang-org/bsc #429: BSV parser should not allow selection by constructor

quark17, 2021-11-03:
The BSV parser allows you write `x.Valid`, using a type constructor as a selector function for that constructor's data, like this:
```
(* synthesize *)
module mkTest ();
  rule r;
    Maybe#(Bit#(16)) x = tagged Nothing;
    $display("Valid value = %0b", x.Valid);
    $finish(0);
  endrule
endmodule
```
If you give BSC the flag `-dparsed`, you can see that the BSV parser has created a pattern-matching case-expression, with only one arm:
```
($display)("Valid value = %0b",
           begin case (x) matches
                 tagged Valid {_theResult__} :  (_theResult__);
                 endcase
           end);
```
This is created by `pPrimaryWithFields` in `src/comp/Parser/BSV/CVParser.lhs`, which allows parsing a constructor name.

When the value doesn't match that constructor, the result of the case-expression is a don't-care due to falling through and not matching any arms (and no default).  This turns into the default don't-care value, 0xAAAA, as seen if we link and run a simulation:
```
Valid value = 1010101010101010
```

My suggestion would be to disallow constructors as selectors, in the parser.  The user can always explicitly write a pattern case-statement, to extract the data, and to explicitly say how to handle the case when the value doesn't match.

Alternatively, the parser could continue to allow a constructor, but instead of a case-expression with no default, it would generate a representation that would trigger an error later (in `typecheck`, say) if the type has any chance of not matching.  For example, BSC already allows the user to write irrefutable match statements:
```
Maybe#(Bit#(16)) x = tagged Nothing;
match tagged Valid .bs = x;
$display("Valid value = %0b", bs);
```
but the `typecheck` stage examines the type and reports an error if isn't a struct or tuple (types with only one constructor, so it's guaranteed to match):
```
Error: "Test2.bsv", line 6, column 18: (T0046)
  Only matching of structs and tuples is supported in this context.
  The following pattern is not supported:
    tagged Valid {bs}
  Consider matching this pattern with a case statement instead.
```
(The check could be loosened to allow tagged unions with only one constructor.)

I will investigate whether anything breaks in the build or testsuite if I remove the parsing of a constructor in `pPrimaryWithFields`.

jahofmann, 2021-11-03:
There is one case where I like to use this construct to reduce boilerplate when I know what type I can expect. For instance, when dealing with "complex" conditions and pattern matching:

```bsv
if(isValid(m) && val > m.Valid || !isValid(m)) begin
      m = tagged Valid val;
end
```

The need for &&& etc in pattern matching makes such constructs a lot less flexible without the direct access notation.

I am aware that isValid is performing the pattern matching internally and that I lose compiler type checking, but I prefer the clarity of this statement over a multiline case or multiple ifs.

quark17, 2021-11-03:
Thanks for the feedback!

Note that the function `validValue` exists, so `validValue(m)` would do the same thing.  One can always write a function to extract the data (and return a don't-care otherwise), but the dot-syntax automatically makes these functions available for all types.  That feels a bit dangerous to me -- I already consider `validValue` to be dangerous -- but it turns out it's part of the language spec: it's specified in Section 9.11.4 (Tagged union member selection) of the BSV Language Reference Guide.  So I don't think I'll be removing it.

FYI, removing this feature doesn't cause any failures in the BSC testsuite, except for the one syntax test that was added to test it specifically test it: `bsc.syntax/bsv05/TypedefStructTaggedUnionSelect.bsv` (in May 2004, when the BSV language was being created).

Also, I was surprised that you said `&&&` is less flexible.  I figured it was more wordy, but I didn't understand in what way it's less flexible -- maybe there's a parser issue there that needs to be fixed?  I tried myself to write the if-condition in one line, without the dot synatx, and I maybe see what you mean, but I do wonder if these are things that can be fixed in the parser and not fundamental limitations.  I did find, though, that if I add an extra set of parentheses, that we can the trinary `?` operator:
```
    if ((m matches tagged Valid .v ? val > v : True)) begin
```
The unnecessary parens could be something to fix in the BSV parser, for example.

For the specific case of `Maybe`, there's also a `fromMaybe` function that can be used to get the value or return a default that's explicitly provided -- so you could do something here like `val > fromMaybe(0,m)`.  But that doesn't generalize to other types (except by writing those functions explicitly).

Anyway, since this is part of the language spec, I won't remove it, I'll just note it as a possible gotcha.  (We might want to insert an error if the undefined branch is taken -- but this might happen dynamically in the run-time of the hardware, so it would need support for dynamic errors, which is something we've wanted to add to BSC.)

closed 2021-11-03 by quark17

jahofmann, 2021-11-03:
Thanks for the alternatives.

I guess flexible is the wrong word to use here, I meant more expressive and shorter in very specific scenarios. In general I like to use the dot-notation when I have multiple matches to different types in a single condition and I don't want to write custom functions for all those types. That doesn't happen too often, but when it does, I am happy that the functionality is available.