B-Lang-org/bsc #435: "Internal Bluespec Compiler Error" after install from the AUR

YuaNFrank, 2021-11-25:
Hi pals, when going through the ICFP2020_Bluespec_Tutoria, after install the bsc from Archlinux user repo, I got this error. Any help? Huge thanks!
```
$ make b_compile 
mkdir  -p build_b_sim
Compiling for Bluesim ...
bsc -u -sim -simdir build_b_sim -bdir build_b_sim -info-dir build_b_sim -keep-fires  -aggressive-conditions  -no-warn-action-shadowing  -check-assert  -cpp +RTS -K128M -RTS  -show-range-conflict      -p src:../Resources:+ -g mkTop  src/Top.bs 
checking package dependencies
Internal Bluespec Compiler Error:
Please report this failure to the BSC developers, by opening a ticket
in the issue database: https://github.com/B-Lang-org/bsc/issues
The following internal error message should be included in your
correspondence along with any other relevant details:
Bluespec Compiler, version 2021.07-22-g61dc0ebb (build 61dc0ebb)
lx: unknown position
make: *** [../Resources/Include_Makefile.mk:100: b_compile] Error 1
```

quark17, 2021-11-25:
This is extremely odd.

Which directory are you running in?  Did you edit the source file in any way?  Can you show the file `src/Top.bs` (relative to where you are executing)?

I don't have Archlinux to test on.  I see that @thotypous is the maintainer of the package, so maybe they can say?  (Also, it's news to me that BSC etc were packaged for this distribution; should we add it to the list on BSC's README?)

YuaNFrank, 2021-11-25:
Thanks for your quick response. Just new to bluespec. Here is the src/Top.bs
```
$ cat src/Top.bs      
-- Copyright (c) 2014-2020 Bluespec, Inc.  All Rights Reserved.

package Top where

mkTop :: Module Empty
mkTop =
  module
    rules
      "rl_print_answer": when True ==> do
          $display "\n\n***** Deep Thought says: Hello, World! *****"
          $display "      And the answer is: %0d (or, in hex: 0x%0h)\n"  42  42
          $finish
```
Thanks a lot! Happy Thanksgiving!

quark17, 2021-11-25:
Can you add `-v` to BSC flags?  You can do this by adding `-v` to the definition of `BSC_COMP_FLAGS` in `../Resources/Include_Makefile.mk`.  Then, when you run `make b_compile`, BSC should display more information about the stages of work that it is doing.  Can you share that output here?

The error you see suggests that something has gone extremely wrong.  For example, the build of BSC may be corrupted; or the GHC compiler used to build BSC may have been corrupted; or you are using a filesystem that is not being properly handled by GHC or BSC.

The error is occuring when trying to read in a file and identify lexical elements.  I can't see how this error can be reached, but maybe the input is possibly not in a format expected by BSC, or something like that?  In most situations (such as file not openable for reading), BSC would give an ordinary error to the user.  This is an internal, unexpected failure.

I don't know enough about Archlinux or how you installed GHC and other dependencies, so I'm unsure how to help.  But showing the `-v` output might give some information.

quark17, 2021-11-25:
Can you run `ghc --version`, to see what version of GHC was used to build BSC?

YuaNFrank, 2021-11-25:
After add -v to the BSC_COMP, 

```
$ make b_compile
mkdir  -p build_b_sim
Compiling for Bluesim ...
bsc -u -sim -simdir build_b_sim -bdir build_b_sim -info-dir build_b_sim -keep-fires  -aggressive-conditions  -no-warn-action-shadowing  -check-assert  -cpp +RTS -K128M -RTS  -show-range-conflict -v      -p src:../Resources:+ -g mkTop  src/Top.bs 
Bluespec Compiler, version 2021.07-22-g61dc0ebb (build 61dc0ebb)
This is free software; for source code and copying conditions, see
https://github.com/B-Lang-org/bsc

Invoking command line:
bsc -u -sim -simdir build_b_sim -bdir build_b_sim -info-dir build_b_sim -keep-fires -aggressive-conditions -no-warn-action-shadowing -check-assert -cpp -show-range-conflict -v -p src:../Resources:+ -g mkTop src/Top.bs

checking package dependencies
starting depend
exec: cc -D__GENC__ -D__BSC__ -E -nostdinc -traditional 459399.c > /tmp/bsc459399.out
starting parsed
Internal Bluespec Compiler Error:
Please report this failure to the BSC developers, by opening a ticket
in the issue database: https://github.com/B-Lang-org/bsc/issues
The following internal error message should be included in your
correspondence along with any other relevant details:
Bluespec Compiler, version 2021.07-22-g61dc0ebb (build 61dc0ebb)
lx: unknown position
make: *** [../Resources/Include_Makefile.mk:100: b_compile] Error 1
```

Probably reinstall ghc?

YuaNFrank, 2021-11-25:
and 
```
$ ghc --version 
The Glorious Glasgow Haskell Compilation System, version 9.0.1
```

quark17, 2021-11-25:
Ah, I wonder if the C-preprocessing is the issue.  If you remove `-cpp` from the Makefile, does it build?

I see that with `-v` BSC has printed this:
```
exec: cc -D__GENC__ -D__BSC__ -E -nostdinc -traditional 459399.c > /tmp/bsc459399.out
```
Is the file `/tmp/bsc459399.out` still around?  Can you view it?

With the `-cpp` flag, BSC is not trying to parse the source file, it is trying to parse the output of `cc -E` which has been written to a temporary file.

YuaNFrank, 2021-11-25:
Oh wow, removing the -cpp flags seems works. and the /tmp/bsc459399.out is not around. So is this problem because of the tmp file are not generated and bsc try to parse it?

Huge thanks.

quark17, 2021-11-25:
This is not a fix, because the `-cpp` flag will be needed by other examples in that repo.

To debug, it would help if you could put the `-cpp` flag back and show the output that gets written to the temp file.

BSC allows source files to contain C preprocessor directives.  See, for example, the directives in `Examples/Eg030b_Bubblesort/src/Bubblesort.bs`.  The `-cpp` flag causes BSC to run `cc` to preprocess the source and write the result to a temp file; then BSC reads in the temp file instead of the original source.

If the temp file is not kept around, you can remove `-u` and add `-dcpp` to cause BSC to dump the result of preprocessing.

quark17, 2021-11-25:
You can also try running the command that BSC has executed.  First, created a file called `test.c` which contains just an import of the source file:
```
#include "src/Top.bs"
```
And then run the command:
```
cc -D__GENC__ -D__BSC__ -E -nostdinc -traditional test.c
```
I expect to see an output like this:
```
# 1 "test.c"
# 1 "<built-in>" 1
# 1 "<built-in>" 3
# 367 "<built-in>" 3
# 1 "<command line>" 1
# 1 "<built-in>" 2
# 1 "test.c" 2
# 1 "./src/Top.bs" 1
-- Copyright (c) 2014-2020 Bluespec, Inc.  All Rights Reserved.

package Top where

mkTop :: Module Empty
mkTop =
  module
    rules
      "rl_print_answer": when True ==> do
          $display "\n\n***** Deep Thought says: Hello, World! *****"
          $display "      And the answer is: %0d (or, in hex: 0x%0h)\n"  42  42
          $finish

# 2 "test.c" 2
```
It may be that the `cc` command is generating something else, possibly even generating an error.

YuaNFrank, 2021-11-26:
Ok I create the file test.c and run the command, then it got
```
$ cc -D__GENC__ -D__BSC__ -E -nostdinc -traditional test.c

# 0 "test.c"
# 0 "<built-in>"
# 0 "<command-line>"
# 1 "test.c"

# 1 "src/Top.bs" 1
-- Copyright (c) 2014-2020 Bluespec, Inc.  All Rights Reserved.

package Top where

mkTop :: Module Empty
mkTop =
  module
    rules
      "rl_print_answer": when True ==> do
          $display "\n\n***** Deep Thought says: Hello, World! *****"
          $display "      And the answer is: %0d (or, in hex: 0x%0h)\n"  42  42
          $finish
# 2 "test.c" 2

```

YuaNFrank, 2021-11-26:
and add the -cpp back and remove -u and add -dcpp give me this:

```
$ make b_compile
mkdir  -p build_b_sim
Compiling for Bluesim ...
bsc -dcpp -sim -simdir build_b_sim -bdir build_b_sim -info-dir build_b_sim -keep-fires  -aggressive-conditions  -no-warn-action-shadowing  -check-assert -cpp +RTS -K128M -RTS  -show-range-conflict -v      -p src:../Resources:+ -g mkTop  src/Top.bs 
Bluespec Compiler, version 2021.07-22-g61dc0ebb (build 61dc0ebb)
This is free software; for source code and copying conditions, see
https://github.com/B-Lang-org/bsc

Invoking command line:
bsc -dcpp -sim -simdir build_b_sim -bdir build_b_sim -info-dir build_b_sim -keep-fires -aggressive-conditions -no-warn-action-shadowing -check-assert -cpp -show-range-conflict -v -p src:../Resources:+ -g mkTop src/Top.bs

starting cpp
exec: cc -D__GENC__ -D__BSC__ -E -nostdinc -traditional 570720.c > /tmp/bsc570720.out
=== cpp:
# 0 "570720.c"
# 0 "<built-in>"
# 0 "<command-line>"
# 1 "570720.c"

# 1 "/home/yuan/Software/ICFP2020_Bluespec_Tutorial/Examples/Eg020a_HelloWorld///src/Top.bs" 1
-- Copyright (c) 2014-2020 Bluespec, Inc.  All Rights Reserved.

package Top where

mkTop :: Module Empty
mkTop =
  module
    rules
      "rl_print_answer": when True ==> do
          $display "\n\n***** Deep Thought says: Hello, World! *****"
          $display "      And the answer is: %0d (or, in hex: 0x%0h)\n"  42  42
          $finish
# 2 "570720.c" 2

-----

 elapsed time: CPU 0.00s, real 0.01s

starting parsed
Internal Bluespec Compiler Error:
Please report this failure to the BSC developers, by opening a ticket
in the issue database: https://github.com/B-Lang-org/bsc/issues
The following internal error message should be included in your
correspondence along with any other relevant details:
Bluespec Compiler, version 2021.07-22-g61dc0ebb (build 61dc0ebb)
lx: unknown position
make: *** [../Resources/Include_Makefile.mk:100: b_compile] Error 1

```

quark17, 2021-11-26:
Thank you.  If I save that output into a file and run BSC on it, I observe the internal error.

It is caused by the `0` in these lines:
```
# 0 "570720.c"
```
BSC attempts to subtract 1 from this value, getting a negative number, which is unexpected.  In the expected output that I attached, the number is 1, so subtracting is OK.  I don't know why your preprocessor is writing 0, but I think BSC should be able to support that.

I believe that BSC is subtracting 1 because of how it reads in these lines: BSC has a rule for newlines (`\n`) that adds 1 to the line count; for lines starting with `#`, BSC reads everything up to the newline, leaving it to the newline rule to apply on the next character; this means that BSC has to pre-emptively subtract 1 to counter the plus 1 that will follow.  I think it would be better for BSC to read in the newline (if it exists and is not EOF) and directly use the linenum provided in the preprocessor directive.  I can look at fixing this.

quark17, 2021-11-26:
This has been fixed in the source.  I assume that someone will need to update the AUR package for it to be fixed there.

Thank you for reporting the bug and your help diagnosing the issue.

closed 2021-11-26 by quark17

YuaNFrank, 2021-11-27:
just rebuild from AUR. It does fix. Thanks!