B-Lang-org/bsc #444: Minor improvements to SystemVerilog rule syntax

thoughtpolice, 2022-01-28:
> Note: I'm not sure if this should be an issue or a discussion. I think you can convert between them. Feel free!

I'd like to make two small suggestions to the rule syntax. I realize this is a topic that can easily [snowball out of control](https://wiki.haskell.org/Wadler's_Law) but I figure I'd throw it out there.

## Shorthand syntax for single-method rules

Briefly, this:

```bluespec
rule r001;
  foo.bar(xyz);
endrule
```

Should be re-writable to:

```bluespec
rule r001 = foo.bar(xyz);
```

The reason for this suggestion is just because many rules tend to be simple connections like this, especially when you're connecting things like top level wires or bus interfaces to other wires. I often have lots of rules like this near my top entity and other top-level hierarchical entities, or when foreign-interfacing with RTL.

I sometimes use the uglier syntax today, which is both less readable and more typing:

```bluespec
rule r001; foo.bar(xyz); endrule
```

## Shorthand for rules which are required to always fire

Similarly, when you're wiring up top-level components like board pins to your design entities, it's incredibly important that many rules are guaranteed to fire on every single cycle.

The need for this is pretty readily obvious since you must drive pins correctly on every clock cycle. And the compiler already provides a mechanism to check for this and tell you when that guarantee can't be made: the combination of `fire_when_enabled, no_implicit_conditions` on the rule:

```bluespec
(* fire_when_enabled, no_implicit_conditions *)
rule r001;
  foo.bar(xyz);
endrule
```

This is common enough that I propose this be re-writeable as:

```bluespec
always_rule r001;
  foo.bar(xyz);
endrule
```

The latter is equivalent to the former with the 2 attributes added. This could also be combined with the previous suggestion:

```bluespec
always_rule r001 = foo.bar(xyz);
```

## Alternatives or other suggestions

None of these are necessary so if they seem superfluous, I understand. But everything helps and I find myself needing to type these quite often.

quark17, 2022-02-04:
Thanks for making suggestions!  (It's also great to hear how people are using BH/BSV and to have discussions about styles and techniques.)  l don't have any sense of how we should use the Discussions area, myself, so I also don't know whether an Issue or Discussion is better.

Some first thoughts I had:

I find that people don't write lots of little connections because they can abstracted it into a single `mkConnection` call.  Are you encountering situations where this can't be done?

I also note that one can write a function like this:
```
module mrule #(Action a) ();
  rule r;
    a;
  endrule
endmodule
```
and then it can be used like this:
```
mrule( foo.bar(xyz) );
```
Does that do some of what you want?  Here, the name is always `r`, because BSV syntax doesn't support constructing rule names from values, but you could write it in BH/Classic to get that:
```
mrule :: String -> Action -> Module Empty
mrule name act =
  module
    rules
      name: when True ==> act
```
Or, replace `True` with a `Bool` argument, to pass in the name, condition, and action.  And you could make a version that has an attribute on the rule, for when you want `always_rule`.

It's also worth noting that rules are first-class objects.  There's a function `addRules` that takes a `Rules` value and adds it to the module, for example, and some functions for composing `Rules` in different ways.  It would certainly be easy to add a function that takes `Rules` and asserts `fire_when_enabled` and `no_implicit_conditions` on each rule.  Adding a new function would have less of a hurdle than changing the syntax, if that's something that's useful to you.

BSV is also verbose in a lot of ways because it was trying to match SystemVerilog.  If there's concern about verbosity, I wonder if switching to BH/Classic (or even some new front-end syntax?) would be on the table?

Also, you say that you run into these a lot.  I'd like to understand more about that.  Interfacing to the outside world (either at the top level or at imported submodules) certainly can be verbose, but I tend to think of those as things you write once, and aren't the majority of what I write.  And that "always" rules are things that I otherwise try to avoid in my designs (sometimes they're a sign of thinking in Verilog rather than in BSV).

thoughtpolice, 2022-02-05:
> I find that people don't write lots of little connections because they can abstracted it into a single mkConnection call. Are you encountering situations where this can't be done?

So a lot of it is when I'm writing things like top-level BSPs or other various doo-dads where you want synchronizers or debouncers or or or... and there isn't always a clean order or mechanism to abstract into a `Connectable` instance for a pin. And like, even if it's not the top-entity for some mega PCIe card with 50+ pins, each half-dozen with their own rules &mdash; lets say just a dozen GPIOs &mdash; for smaller designs it's a much better use of my "mental bandwidth" to just write the design out in a first-order way with little complication. So while there might be alternatives, it's often just an easily understandable, straight-forward alternative.

However, the very act of writing the components themselves are something to consider too! A good example is a small prescaler module I wrote; you definitely *want* that module to fire its counter-decrement rule on every clock cycle, so you'd use `always_rule` there (because you want it to match the bus for an I2C peripheral or whatever). Yes, other people won't need it later when they use the module, but I tend up writing a lot of small reusable modules with lots of little rules &mdash; because that's what bsc excels at! In this sense, it's a victim-of-your-own-success/hoisted-by-your-own-petard kind of thing...

That said my design is in the early stages so in the long run the relative weight of these rules vs others will probably go down. But again, this is mostly just a case of ergonomics.

> module mrule #(Action a) ();

This is actually a good alternative I hadn't thought about! In general I don't write tons of abstractions around rules (though I have the occasional use for things like `ModuleContext`) so this and `addRules` etc had slipped my mind. (For modules themselves I generally prefer tend to be somewhat first-order, so that synthesis boundaries are more well defined, but for rules this doesn't apply and higher-order constructions seem fine.)

The note about rule names for BS Classic being `String` values that can be set also answers a question I had which is "Why the hell are rule names double-quoted in `.bs` files, and not in `.bsv` files." So that's good to know!

> BSV is also verbose in a lot of ways because it was trying to match SystemVerilog. If there's concern about verbosity, I wonder if switching to BH/Classic (or even some new front-end syntax?) would be on the table?

Honestly, I have had "move over to the Haskell syntax" on my TODO list but the thing is... there's a decent Visual Studio Code extension for `.bsv` syntax, and not one for the Haskell syntax. And the SystemVerilog syntax has somewhat grown on me as a result; I don't find its verbosity to be particularly limiting (the limiting factor is my own brain power/mental bandwidth/skills at RTL!) It also wouldn't hurt if the converter utilities between sv/classic syntax were more polished and emitted "prettier" converted code. That said, I'm not one to be held back by that, but...

A thing I've wanted to do for *a while* now is write a **[tree-sitter](https://tree-sitter.github.io)** grammar for the Bluespec Classic and SV syntaxes. The reason for this is that such a grammar can be relatively easily plugged into editors like Atom/VS Code for basic outlining/syntax highlighting purposes. Then it would be easier for me to switch while not losing my nice highlighting. It would also open the way for Bluespec to be better integrated into GitHub code search...

Anyway, yes, this is certainly going to happen down the road, though clearly I don't quite need it to address my issue here just yet...

> Also, you say that you run into these a lot. I'd like to understand more about that. Interfacing to the outside world (either at the top level or at imported submodules) certainly can be verbose, [...]

So I think it's worth keeping in mind that the scale of designs here is, I suppose, not of a normal distribution. If you are working on a large bluespec codebase, then over time the relative proportion of always_fire/always_enabled rules vs "normal" (lets say unannotated or otherwise "normal" rules) will become very very small, I suspect. Part of this is because both these rules constrain the available schedules but also because, well, you just don't need them everywhere.

But I also write lots of designs that are like, 50-75 lines! For small demos, testing small devices or something on a breadboard, or whatnot. I take that, with the small top entity designed to mimic a basic BSP, then throw that into Questa or Icarus or Verilator or whatever.

Now again part of this can be solved by more general re-use. Like again, if you have a prescaler available, you can use it, so you wouldn't write that stuff yourself, it's abstracted. But at some level I also find you can spend *too much time* on stuff like that. So for those cases I think it's easier to just do it "The Verilog Way" &mdash; for example, just wire some input method ports to a pin, use an always-firing rule to wire the pin to a debouncer or a synchronizer in the stdlib (or even just two `Reg`s if you're being really lazy and old-school &mdash; I actually do this because it would be trivially understandable for anyone) and call it a day.

So in some sense I want to be clear I don't think of this sort of thing as an endorsement for "Write lots of rules that always fire" or "Write things The Verilog Way" or whatever; it's mostly just making something that already had to be written a bit easier to write. I think those cases matter a bit too!

But with the `mrule` suggestion above, I might be satisfied...

thoughtpolice, 2022-02-05:
> So in some sense I want to be clear I don't think of this sort of thing as an endorsement for "Write lots of rules that always fire" or "Write things The Verilog Way" or whatever; it's mostly just [...]

I'll also note that there is a broader discussion that could be had here about language design; in short, I think the annotation/rule distiction is, more broadly, a small wart in retrospect.

I don't think this topic should be derailed, but I think, as an example, that the fact that annotations like `fire_when_enabled` are so semantically influential on the meaning of a rule perhaps suggests they should actually *not* be annotations, but part of the direct user-exposed syntax. This is just a wart in the syntax/semantics cohesion of the language as it stands, IMO. If the annotation is critical to the actual meaning of the circuit, it feels important enough to be in the language. What would it look like? I don't know. Someone else can figure that out.

OTOH, you could argue that because Bluespec designs are more like specifications (and thus you're free to imagine many circuits that might perform your desired task), that the separation of this into an annotation means that some other implementation could interpret a rule in some other, semantically meaningful way, without being inhibited by needing to support extra semantics, etc. If I had to guess based on my experience designing Haskell-ish languages (guess which one), that would be my bet for the motivation: that the separation was intended to more reflect the separate compiler implementation specifics from the language semantics.

But IME, time proved that the language is generally advanced enough that fresh, source-compatible re-implementations are relatively rare, difficult, and most importantly &mdash; the design space is rich enough any serious attempts of this nature will be lured by the siren song to try new, incompatible ideas (for good cause!) And so the syntax and semantics of the existing tool become married together, anyway, because they never get cleanly separated. This was, and remains, a majorly vocal complaint about Glasgow Haskell: that yes, it's true many extensions it has may inhibit other kinds of new implementations of "Standard Haskell", should it acquire those features. But on the other hand the language was already difficult to implement even with mild extension support, and anyone who would have tried that would have likely decided to "go beyond" that and just push into new (incompatible) territory anyway. So why even bother with all the annotations, etc? A richer source syntax would be better, IMO, but that's just me.

So I'm rambling at this point but I suppose my point is, I think a big part of *why* this separation even occurred in the first place is probably important to think about when discussing something like this. I realize I'm speculating the motivations a bit here, they might be lost to history, but it's based on my similar motivations in a Haskell implementation. And perhaps if anyone out there reads this issue 10 years from now and is designing a new HDL in the cloak of Bluespec, that might be a lesson to take away in terms of design quirks...