B-Lang-org/bsc #446: `notEmpty` signal in `mkSizedBypassFIFOF` not set correctly.

pawks, 2022-01-31:
The `notEmpty` signal in `mkSizedBypassFIFOF` does not read `True` in the cycle where the fifo is empty and an element is being enqueued. Thus using the `notEmpty` signal is used as a condition to perform other actions(which conditionally performs a dequeue) causes the fifo to behave as a normal FIFOF instead of a bypass FIFOF.

quark17, 2022-02-02:
Thank you for submitting a PR.  Do you happen to have a testbench that can be added to the testsuite?  Tests for the `SpecialFIFOs` package are in `testsuite/bsc.lib/fifo/` and I see that `TbBypassFIFO.bsv` has a test for the enq/deq/first methods of `mkSizedBypassFIFOF`.

That test also has a comment which notes that the `notEmpty` method is not the same as the ready signal for the `deq` method.  This is the issue that you're pointing out, and it seems that we became aware of it in 2014 (when the test was written) but didn't take any action at the time.

In theory, a BypassFIFO can have three different `notEmpty` methods: one that gives the status before `enq` or `deq` has been called; one that gives the status after `enq` has been called but before `deq` has been called; and one that gives the status after `deq` has been called.  All three of these are valid definitions, and you could imagine a FIFO that provides all three.  The `FIFOF` interface has only one method, though, so it should probably be defined as the version that people most expect/need.  I would guess that most people expect it to match the ready condition for `deq` (the status after `enq` but before `deq`), but I'm not sure if that's the version that is most needed for designs.

Probably the reason that we took no action in 2014 was because we had just received blowback from a customer whose designs broke when we changed the `mkBypassFIFO` module.  In that situation, the `notEmpty` signal wasn't either of the three valid definitions; it was a buggy version that could accidentally schedule in the wrong place and provide a stale value, given its location in the schedule relative to the `enq` and `deq` methods.  In that case, we fixed the bug, but also had to choose which of the three points it should schedule at.  The customer was very upset about backwards incompatible changes, and so we were probably overly cautious about making changes like that again.

In this case, there's no bug in the `notEmpty` method; it's just defined as the status at the start of the clock cycle (before `enq`).  There are valid reasons why a design might want to access the status at that point, and not after the enqueue.  With your change, someone wouldn't be able to write that design.  Although I understand that, without the change, you can't write your version.

Instead of your current PR that replaces the current behavior with the new behavior, I would prefer a PR that allows designs of both kinds to be written.  (Separately, we can consider whether the name `mkSizedBypassFIFOF` should continue to provide the current behavior or different behavior.)

For example, we can define a new type `FIFOFF` (or some better name), that has `notEmpty0` and `notEmpty1` (and perhaps even `notEmpty2`) methods.  (And the same for `notFull`.)  The current `mkSizedBypassFIFOF` can be renamed `mkSizedBypassFIFOFF` and adjusted to return the `FIFOFF` interface, defining all the variations of the methods.  This module could be instantiated directly, but we would also define wrapper modules for it that return `FIFOF` interfaces, each picking one of the method versions to use.  The name `mkSizedBypassFIFOF` can be used for a wrapper module that preserves the current behavior by exposing just the `notEmpty0` method (for backwards compatibility) and a new module `mkSizedBypassFIFOF2` (or some better name) could be defined to return a `FIFOF` interface using `notEmpty1`.  Then, in your design where you are currently using `mkSizedBypassFIFOF`, you could instead instantiate this other version.

This would allow all users to write their kinds of designs.  And separately we can consider whether the name `mkSizedBypassFIFOF` should be repurposed for one of the other meanings (requiring current users to instantiate a different module name).

neelgala, 2022-02-03:
@quark17 shouldn't the notEmpty/Full signals of `mkSizedBypassFIFOF` have the same meaning as `mkBypassFIFOF` - purely from the naming terminology of the modules. I believe this PR achieves that.

I agree with your proposal for `mkSizedBypassFIFOFF` and the new type `FIFOFF` - it will surely add value.

quark17, 2022-02-16:
I will accept the PR, but I'd like to have a test to commit with it.  Unless you're working on something, I'll see if I can write a test.

I also didn't want to commit the change just before tagging a release.  Now that the release is done, we can merge and have plenty of time before the next release for someone to complain if compatibility issues are encountered.

quark17, 2022-03-05:
I made a test for the schedule of `mkSizedBypassFIFOF` and found that the `notFull` and `notEmpty` methods were allowed to be called at points in the schedule where the value is invalid!  I have submitted the test and (hopefully) a fix in PR #462.

This is only testing the schedule.  It isn't testing that `notEmpty` is defined properly for where it's placed in the schedule.  We probably still want to add a test for that, or update `bsc.lib/fifo/TbBypassFIFO.bsv` (the test for `mkSizedBypassFIFOF`, despite not saying so in the name) to include a test for it.