B-Lang-org/bsc #457: Cannot build Bluespec Compiler with "BSC_BUILD=DEBUG" option

ssmolov, 2022-02-18:
I try to build BSC in a debug mode. To accomplish that, I use the following command:
```
make PREFIX=/home/<username>/bsc BSC_BUILD=DEBUG release
```
Build crashes with the following confusing error log:
```
*** End of Offense ***

<no location info>: error: 
Compilation had errors

make[2]: *** [Makefile:344: bluetcl] Error 1
```
I've installed tcl-dev 8.6.9 by the way.

quark17, 2022-02-19:
I observe the same behavior on macOS (11.6.4 x86) and Debian (10.11).  (Setting `PREFIX` is not required for triggering the bug.  Nor is using the `release` target specifically.)  The problem occurs when generating `bluetcl` in `src/comp/` and there is a very large error message that precedes the "no location info" message:
```
[146 of 151] Compiling HTcl             ( ../vendor/htcl/HTcl.hs, /Users/quark/Work/git-repos/blang/bsc/src/comp/../../build/comp/HTcl.d_o )
*** Core Lint errors : in result of Simplifier ***
../vendor/htcl/HTcl.hs:629:11: warning:
    Out of scope: $s$wworker_sp59
                    :: HTclCmdGrammar
                       -> HTclCmdGrammar
                       -> [ForeignPtr XTclObj]
                       -> [([Char], HTclCmdElem)]
                       -> IO ([(String, HTclCmdElem)], [TclObj], HTclCmdGrammar)
                  [LclId, Arity=5, Str=<L,U><L,U><S,U><L,U>]
    In the RHS of htclMatchGrammar :: TclInterp
                                      -> [TclObj]
                                      -> HTclCmdGrammar
                                      -> IO ([(String, HTclCmdElem)], [TclObj], HTclCmdGrammar)
    In the body of lambda with binder interp_ahAO :: TclInterp
    In the body of lambda with binder objs_ahAP :: [TclObj]
    In the body of lambda with binder cmd_grammar_ahAQ :: HTclCmdGrammar
    In the RHS of $wworker_snCs :: [(String, HTclCmdElem)]
                                   -> [TclObj]
                                   -> HTclCmdGrammar
                                   -> IO ([(String, HTclCmdElem)], [TclObj], HTclCmdGrammar)
    Substitution: [TCvSubst
                     In scope: InScope { ... }
                     Type env: []
                     Co env: []]
*** Offending Program ***
...
*** End of Offense ***
```
This looks like it could be a GHC bug?  I was using GHC 8.10.7 on macOS and 8.4.4 on Debian.  I will try setting up GHC 9.x and seeing if it still has this issue.  Otherwise, it may be worth opening a bug in the GHC repo and seeing what they suggest.

quark17, 2022-02-19:
I used GHC 9.0.1 on macOS and that successfully builds.  So you might try updating to a newer GHC.  I used `ghcup` to install a specific version, and the instructions for that are in the build/install instructions for BSC.

For older GHC, a way to avoid triggering the bug is to reduce the optimization level.  I notice that the `Makefile` in `src/comp/` gives the flag `-O2` to GHC.  You can reduce this (to `-O1` or `-O0`) by setting `GHCOPTLEVEL=` in the `make` command.  On Debian with the older GHC 8.4.4, I was able to successfully build bluetcl by reducing the level to one:
```
make GHCOPTLEVEL=-O1 BSC_BUILD=DEBUG ...
```
We could edit the `Makefile` to automatically reduce the optimization level when the GHC version is earlier than 9.x and debug build is selected.

Incidentally, even though it succeeds, there are a ton of "Core Lint" warnings in `IExpandUtils` and `IExpand`:
```
IExpandUtils.hs:2672:1: warning:
    INLINE binder is (non-rule) loop breaker: inferName

IExpandUtils.hs:2607:1: warning:
    INLINE binder is (non-rule) loop breaker: toHeap

IExpandUtils.hs:2629:1: warning:
    INLINE binder is (non-rule) loop breaker: toHeapCon

IExpand.hs:2721:1: warning:
    INLINE binder is (non-rule) loop breaker: evalStaticOp'
```

closed 2023-07-29 by quark17