B-Lang-org/bsc #458: BVI for arrays

podhrmic, 2022-03-01:
Hello! I have the following situation:

A verilog module with these ports:
```verilog
module Generate_Sensor_Trips
    #(localparam NChannels = 3)
    ( input logic [NChannels * 32 - 1:0] vals,
      input logic [NChannels * 32 - 1:0] setpoints,
      output logic [NChannels - 1:0] out
    );
```

The input logic (`vals`,`setpoints`) is effectively an array of 3 values of 32bits, I would like to write in BSV the following:
```
interface SensorTrips_IFC;
    (* always_ready *)
    method Bit#(3) generate_sensor_trips (Bit#(32) vals[], Bit#(32) setpoints[]);
endinterface
```

However, when compiling with this interface, the Bluespec compiler then complains that:
```
The provisos for this expression could not be resolved because there are no
  instances of the form:
    Bits#(Array#(Bit#(32)), a__
```

Defining the interface as:
```
interface SensorTrips_IFC;
    (* always_ready *)
    method Bit#(3) generate_sensor_trips (Bit#(96) vals, Bit#(96) setpoints);
endinterface
```
works, but it is not pretty, and it is not immediately clear that the module expects 3 32-bit int values, rather than one 96-bit integer.

What do I need to add to `provisos` to be able to connect the ports as I described?

rsnikhil, 2022-03-01:
The 'Array' type is not in the 'Bits' typeclass (see 'Type Classes for Array' table in the Reference Guide section 'B.2.13 Array').  In part the reason is, because the size is not part of the type, the size cannot always be resolved statically (which is needed to generate hardware).

Thus, 'Array' types cannot be used as run-time values (in registers, on wires, FIFOs, BRAMs, etc.).  The error message you cite is reporting this.

However, the 'Vector' type _is_ in the 'Bits' typeclass (see 'Type Classes for Vector' table in the Reference Guide section 'C.3 Aggregation: Vectors').  Its size is part of its type.

Thus, 'Vector' types can be used for run-time values.

Your interface can be written using Vectors:

    import Vector :: *;

    interface SensorTrips_IFC;
       (* always_ready *)
       method Bit#(3) generate_sensor_trips (Vector #(3, Bit #(32)) vals,
                                             Vector #(3, Bit #(32)) setpoints);
    endinterface

In general I use Vectors whenever possible, and use Arrays only rarely (e.g., the interface type of CRegs is given as an Array).

Note, the size type-parameter for Vectors (Vector #(size_t, ...)) can be a polymorphic type, so you can write functions and modules that are polymorphic in the Vector size.

podhrmic, 2022-03-01:
Thanks for the explanation, using vectors makes it much nicer!

closed 2022-03-01 by podhrmic