B-Lang-org/bsc #464: Docs: clarify that BRAM modules can do 1-cycle operations

danderson, 2022-03-31:
This is a minor suggestion for improving the library reference documentation. I'm happy to send a PR to implement it, but first I wanted to make sure my proposal makes sense.

I'm still a relative newbie to BSV and the libraries. Reading the documentation for the BRAM library, it wasn't clear to me at a glance that the modules support 1-cycle accesses. In particular, the wording of the docs for the `BRAM_Configure` struct implied to me that BRAM modules had a minimum access time of 3 cycles, because `latency` can be 1 or 2 depending on data registration, and then `outFIFODepth` adds another 2 to that.

What I'd missed, of course, is that the BRAM's output FIFO is a `BypassFIFO`, so as long as the reader keeps up, the FIFO depth is irrelevant and the latency is dictated purely by whether I choose to register the output data.

I was further confused because the `BRAM_Configure` struct has another field that mentions bypassing, `allowWriteResponseBypass`. So, naively, I assumed that "bypass FIFO" was in reference to this write bypassing behavior.

I'm sure none of this is confusing to a more experienced hardware designer, especially one already familiar enough with BRAM design, or intimate enough with the Bluespec libraries to recognize that BypassFIFO is a reference to the SpecialFIFOs package. But it took me more time than I'd like to admit, and digging into BRAM's source code, to convince myself of the actual behavior.

My proposal would be to edit the BRAM library documentation slightly:
 - Explicitly state that the `latency` field of `BRAM_Configure` completely defines the access latency, and `outFIFODepth` only matters when faced with slow readers
 - Cross-reference mentions of `BypassFIFO` to the `SpecialFIFOs` package, to make it more obvious that an unfamiliar reader should skip forward in the reference doc and go understand what a `BypassFIFO` does (this is admittedly a small lie, because BRAM privately implements its own bypass FIFO, I assume for historical reasons - but afaict, the semantics are identical to the bypass FIFOs in `SpecialFIFOs`)

Do you think these changes would make sense?

quark17, 2022-04-08:
Thank you!  And, yes, this looks good.

I agree that it's confusing that the "bypass" in "BypassFIFO" is unrelated to the BRAM's write bypass feature (and that should probably be pointed out explicitly in the documentation).  As you discovered, the term "BypassFIFO" is a convention that developed at Bluespec to name a certain kind of FIFO, and newcomers may have not have learned this meaning yet, so I agree that it would be good to add a note referring readers to `SpecialFIFOs` for an explanation of "Bypass FIFO".

And I see what you mean about it being a small lie.  The `BRAM` package defines an adapter function, that takes an ordinary FIFO and turns it into a BypassFIFO.  Separate from updating the documentation, it might be good if someone wanted to move the definition of this adapter function into the `SpecialFIFOs` package -- or even update the `BRAM` modules to not use an adapter!  The `BRAM` modules instantiate an ordinary Sized FIFO and then apply the adapter to make it a Bypass FIFO, where they could have just instantiated `mkSizedBypassFIFO` in one go; I'd guess they are written this way because `mkSizedBypassFIFO` didn't exist in `SpecialFIFOs` at the time.  (Several projects inside Bluespec were independently evolving their own BRAM and FIFO libraries.  The current `BRAM` package came out of unifying those, but the `SpecialFIFOs` package didn't get `mkSizedBypassFIFO` until later.)

danderson, 2022-04-08:
Sounds good! I started with a PR to make BRAM use SpecialFIFOs, then I'll give a shot at updating the docs.