B-Lang-org/bsc #465: why there is no source and destination synchronization reset in syncfifo and syncreg. they are simply connected via wire.

Kungfu-ckoder, 2022-04-02:


Kungfu-ckoder, 2022-04-02:
thanks in advance

quark17, 2022-04-08:
GitHub issues have a title and a description.  You seem to have a left the description field blank and used the title field to explain the problem.  In the future, can you enter a short title and use the description field to provide a longer explanation?

If I understand, you're noting that the modules `SyncFIFO.v` and `SyncRegister.v` take in a source domain reset (`sRst`) and no destination domain reset, and they use the source domain reset unchanged in the destination domain.  In `SyncRegister.v` that's in this always block:
```
always @(posedge dCLK or `BSV_RESET_EDGE sRST)
   begin
      if (sRST == `BSV_RESET_VALUE)
        begin
           dD_OUT <= `BSV_ASSIGNMENT_DELAY init ;
        end // if (sRST == `BSV_RESET_VALUE)
      else
```
and in `SyncFIFO.v`, it is doing the same thing, but it has created a signal named `dRst` to use (even though it's just assigned the `sRst` value):
```
assign dRST = sRST ;

always @(posedge dCLK  or `BSV_RESET_EDGE dRST)
  begin
     if (dRST == `BSV_RESET_VALUE)
```
And you're asking why `sRst` is used directly in these always blocks, instead of going through some synchronizer that creates a different reset signal, I think?

I am not an expert on this, so I will need to ask the person who wrote these modules.  But I would ask you: what would it mean to you to synchronize `sRst`?  If you look at `SyncResetA.v`, you will see a recommended synchronizer for resets, and you'll see that the output reset signal asserts immediately (asynchronously) and the synchronizer just holds the assertion until a destination clock edge so that it can be de-asserted synchronously.  This is implemented in the same way that `SyncRegister.v` and `SyncFIFO.v` are using the reset:
```
always @( posedge CLK or `BSV_RESET_EDGE IN_RST )
  begin
     if (IN_RST == `BSV_RESET_VALUE)
```
So I would guess that no synchronizer is needed, because the always blocks in `SyncRegister.v` and `SyncFIFO.v` only need to consider the asynchronous assertion of the reset, because they are reading values directly from the source domain.  A synchronous deassertion would only matter for logic that's another step further into the destination domain, reading only values in the destination domain.