B-Lang-org/bsc #466: UngatedClockMux.v isn't a glitch free circuit. using it in cdc designes can lead to serious problems

Kungfu-ckoder, 2022-04-02:


Kungfu-ckoder, 2022-04-02:
thanks in advance

quark17, 2022-04-08:
BSC's `Clocks` library provides (among many others) four modules for selecting clocks: `mkClockMux` and `mkClockSelect` (where the clocks have both an oscillator and a gate signal) and `mkUngatedClockMux` and `mkUngatedClockSelect` (where the clocks have only an oscillator signal).  These source-level modules are implemented as wrappers around Verilog primitives, which are the files `ClockMux.v`, `ClockSelect.v`, `UngatedClockMux.v`, and `UngatedClockSelect.v` (in the `src/Verilog/` directory).

You noted that the `Mux` versions aren't glitch free.  That is correct.  This is documented in library guide (see `doc/libraries_ref_guide/LibDoc/Clocks.tex`):

> The `mkClockMux` module is a simple combinational multiplexor with a registered clock selection signal, which selects between clock inputs `aClk` and `bClk`.  The provided Verilog module does not provide any glitch detection or removal logic;  it is the responsibility of the user to provide additional logic to provide glitch-free behavior.

> The `mkClockSelect` module is a clock multiplexor containing additional logic which generates a reset whenever a new clock is selected.

I am not an expert on this, but I also suspect that you probably only want to use these modules if they make sense for your target hardware and synthesis tools.  I don't have a sense of how Vivado would handle this for Xilinx FPGAs, for example, so I might prefer to use a Xilinx macro instead.

closed 2022-05-09 by quark17