B-Lang-org/bsc #486: Drop support for macOS 10.15

quark17, 2022-07-19:
The macOS environments that we support (and provide tar-file releases for) are limited to the ones that we have access to test on, which is currently the VMs that GitHub provides for Actions: macOS 10.15, macOS 11, and macOS 12 (which we will add shortly).  GitHub has already started deprecating the macOS 10.15 Actions runner image (since 31 May) and it will be fully unsupported by 30 August 2022.

We will be tagging a 2022.07 release this month, and will provide a macOS 10.15 tar-file for it.  But then shortly after, we will remove the macOS 10.15 CI.  If support is still important to anyone, and you can provide access to a VM (for GitHub CI to run on), then let us know.  Otherwise, expect the support to end in August.

ThePerfectComputer, 2022-10-25:
Well, I use BSC on 10.13 and it seems to work fine. I suppose I can create a pull request if it breaks.

kenta2, 2022-11-30:
For reference, this seems to be official thread about 10.15 ending: https://github.com/actions/runner-images/issues/5583

closed 2023-08-24 by quark17