B-Lang-org/bsc #489: Float to fixed of a denormalized number

michael-roe, 2022-07-20:
https://github.com/B-Lang-org/bsc/blob/main/src/Libraries/Base3-Math/FloatingPoint.bsv,
line 3156 says:

// todo: does this work for subnormal?

it seems that, no, it does not work for subnormal floating point numbers.

This ends up being an issue for us with CPUs written in Bluespec, because the failure to handle this case shows
up in randomized testing of the CPU.

quark17, 2022-07-26:
Thank you for reporting this.  We are exploring the FloatingPoint library on a set of tests from IBM for IEEE 754R ([here](https://research.ibm.com/haifa/projects/verification/fpgen/test_suite_download.shtml)), but it doesn't include tests for converting from float to fixed, so it wouldn't have caught this issue.  If you have examples that are failing and can provide the inputs and expected outputs, for testing, that would be helpful.  If you're able to submit a fix, that would be wonderful too.

michael-roe, 2022-07-26:
The example of it failing is when a negative subnormal number is rounded down (rounding mode is towards negative infinity), it should be rounded down to -1, but the current code rounds it up to 0.

michael-roe, 2022-07-26:
P.S. I am also running IBM's FPGen tests. I am running them on the Flute RISCV soft core; as this is written in Bluespec, it indirectly tests the Bluespec floating point libraries.

The float-to-int bug was found using a different randomized tester.