B-Lang-org/bsc #504: Tcl-X11 on MacOS

ThePerfectComputer, 2022-10-12:
I've built a version of tcl that links against x11 instead of quartz for MacOS so that I can get BDW working.

I deleted these [two lines](https://github.com/B-Lang-org/bsc/blob/main/platform.sh#L75-L76)
as I now have `which tclsh` point to the x11 install.

But when I run make however, and then run the subsequently built bluetcl, bluetcl points out that its linked against the MacOS tclsh...

Are there more tcl flags I need to change through the build scripts in bsc sources?

quark17, 2022-10-12:
> I deleted these [two lines](https://github.com/B-Lang-org/bsc/blob/main/platform.sh#L75-L76)

Further down in that file, it's constructing the comple/link flags for tcl, and you probably have to update those as well.  See, for example, lines 94-98 and lines 136-140, which have the same comment about avoiding Homebrew's version on macOS.

You may also need to tell Homebrew to install Itk, for its tcl/tk.

I develop on a mac, so normally I could help, but I think my files might need some refreshing (possibly I updated the OS recently and didn't reinstall stuff), because I'm getting this with the default tclsh on my mac (which is at 11.7):
```
% package require Itk
macOS 11 (1107) or later required, have instead 11 (1106) !
```

kammoh, 2022-10-12:
I'd like to add that the system `tcl` which comes with recent versions of macOS is not only very outdated, but broken in many ways and unusable for both `bluetcl` and `bdw`. Homebrew's `tcl-tk` is mostly usable, by just removing the current "workaround", and some other minor modifications:

https://github.com/kammoh/bdw/tree/macos-tcl
https://github.com/kammoh/bsc/tree/macos-tcl

I have only tested this on macOS 11.x/x64, 12.x/arm64, and 13.0/arm64.

Unfortunately, the `graphviz` package on homebrew does not come with TCL support, so you have to modify the recipe, or manually install `graphviz` with `--enable-tcl=yes --with-tclsh=$(brew --prefix)/opt/tcl-tk/bin/tclsh`

ThePerfectComputer, 2022-10-14:
I was able to start btw - and things looked fine - but it seems that the tcl that comes with home-brew uses itch 4+ where btw want 3.2... This really causes some usability issues. Installing 3.2 on MacOS is complicated to say least...

quark17, 2022-10-20:
> but it seems that the tcl that comes with home-brew uses itch 4+ where btw want 3.2...

When you say that BDW wants 3.2, is it just because of the `package require` statements at the top of `src/iwidgets4.0.1/iwidgets.tcl`? In the modifications that @kammoh linked to, they just removed the version numbers from the `require` statements.  If you do that, does BDW run fine?  Or does it actually contain Itcl calls that don't work with version 4+?

@kammoh, thank you for the comments!  We definitely should move away from using the Tcl that comes with macos -- it even prints a big warning when you use it, saying that it only exists for legacy software and may be removed in future macos releases.

If we do make this change, does it still make sense to provide a pre-built macos tarball?  I would think not, since we can't be sure what version of tcl the user will have, and I assume that bluetcl could have problems if used with a different version?  We could provide a tarball and say what version it requires, but I'm not sure if that's useful?  It might be best to stop providing pre-built tarballs for macos and, instead, to provide a Homebrew formula for it, so that people can install it via the package manager, which can know to rebuild bsc when its dependencies change.

I don't know anything about providing your own formula or tap for Homebrew, so I will have to figure that out.  At the very least, it looks like we can create a new repo (called `homebrew-b-lang-org`?) that can be a tap; but I'd be curious to know if we could just provide the formula here in this repo.  Has anyone made a Homebrew formula for recent BSC?  I assume that it shouldn't be hard -- I see that there's a tap and formula from 2020 ([here](https://github.com/willjwon/homebrew-bsc)) that could be a start, but I think it might need some fixing in how it installs (the choice of `PREFIX` coupled with BSC's use of `lib` as a subdirectory name, might mean that everything in BSC's `lib` directory will get symlinked into `/usr/local/lib/` maybe?).

quark17, 2022-10-21:
@kammoh I was able to build BDW using your approach.  (Because tcl-tk is built keg-only, I had to add it to `PATH` and `PKG_CONFIG_PATH`.)  It started up fine, but when I tried New or Open from the Project menu, I got an error:
```
bad option "-minsize": must be -defaultextension, -filetypes, -initialdir, -initialfile, -message, -parent, -title, -typevariable, -command, or -confirmoverwrite
bad option "-minsize": must be -defaultextension, -filetypes, -initialdir, -initialfile, -message, -parent, -title, -typevariable, -command, or -confirmoverwrite
    while executing
"tk_getSaveFile -minsize "350 200" -title "New Project"  -filetypes $type -initialfile $initfile"
    (procedure "menu_actions::project_new" line 11)
    invoked from within
"menu_actions::project_new"
    (menu invoke)
```