B-Lang-org/bsc #506: Bluespec Compiler Sources

ThePerfectComputer, 2022-10-22:
Is there any material that goes into how the Bluespec compiler works/is structured at a high level?
I read in the [Riscv OOO Bluespec Paper](https://github.com/bluespec/Toooba/raw/master/Doc/micro2018.pdf) that the compiler constructs a conflict matrix for example...

Where could I find out more about this?

quark17, 2022-10-23:
Users of BSC should definitely have a model of how BSC is doing scheduling and the kind of schedules that it generates.  Some might want to know more detail about how exactly BSC does it.  I'm not sure which you're asking for, so I'll mention a few things off the top of my head.

There is a book _BSV by Example_ (available as PDF [here](https://github.com/BSVLang/Main/blob/master/Tutorials/BSV_by_Example_Book/bsv_by_example.pdf), in the BSVLang GitHub account, which will eventually be merged with this one; the book is also available as a print-on-demand paperback).  Chapter 7 is on scheduling.  I would guess that the BSV Reference Guide and the BSC User Guide would have some info on scheduling.  But I don't recall how well any of these do at presenting a complete model of what BSC's scheduler is doing.  (And, if not, that's certainly a document we should have.)  I think the User Guide might say something about the WILL_FIRE and CAN_FIRE signals that are generated in the output Verilog, which is an important part of the intuition.

More detail would certainly be available in some of the papers that Bluespec Inc wrote, on scheduling.  However I don't think these were published.  I see a draft of a paper on "Esposito Scheduling", to be submitted for ICCAD 2005 (adapted from a draft for DAC 2005) but I don't know that it was published.  I also see an incomplete draft of a paper for DAC 2009, on a later scheduling improvement: constructing separate "execution order" and "urgency order" graphs and then combining them into a final unified graph.  I imagine that we could make these public and/or massage them into a new document on the details of BSC's scheduling.

There are some public documents (and documents by other people) that could give some insights.  For example, there was a patent for "Esposito Scheduling" and you can read it [here](https://patents.google.com/patent/US7647567B1/en) but it's likely written in patent-ese and may take some deciphering.  The basis of BSC's scheduling is that primitive modules come with "annotations" about the scheduling relationships of the methods, and when synthesizing a user-written module, BSC uses this info to schedule rules (and to deduce the annotations for the new module).  This concept of annotations is discussed in papers at MIT (which may not exactly describe what BSC does, but is similar).  Examples include [Modular Scheduling of Guarded Atomic Actions](http://csg.csail.mit.edu/pubs/memos/Memo-468new/memo468.pdf) (DAC 2004, by Rosenband and Arvind) and [Scheduling as Rule Composition](https://people.csail.mit.edu/ndave/Research/scheduling_07.pdf) (MEMOCODE 2007, by Dave, Arvind, and Pellauer).

But maybe the best answer is that the code for BSC's scheduler is heavily commented.  So you might try reading the code.  The function `aSchedule` in `src/comp/ASchedule.hs` is one long do-block that sequentially goes through all the steps in scheduling.  (It was broken into a few pieces, because GHC had trouble with a large do-block at one point.  They're called `aSchedule_step1`, `aSchedule_step2`, and `aSchedule_step3`, but they can still be read as one long sequence.)  Each step in the sequence is often a function call -- the definitions are later in the file and often good comments on them, explaining what they do.

You'll see, for example, that the code constructs three "conflict graphs": CF (conflict-free), SC (sequentially composable), PC (parallel composable).  The nodes in these graphs are rules, and edges exist to express whether the pair of rules are conflict-free or composable.  The graphs are initially populated by finding all the methods that two rules call on the same submodules and then consulting the annotiations (like CF, SC, PC) for those pairs of methods.  But later info can be added to graph, as BSC determines that rules conflict because of resource constraints (say) and BSC has to introduce conflicts.  BSC also looks at mutual exclusion of the predicates, and adds that info into the mix.  Eventually, BSC creates two graphs: an execution order graph and an urgency order graph.  If there are any loops in these graphs, BSC deals with them (often by reporting an error to the user).  Finally, these graphs are combined into a unified scheduling graph -- there is a node for each rule's predicate ("Sched_<rulename>") and a node for each rule's body ("Exec_<rulename>") and edges for ordering dependencies.  For example: the predicate has to be computed before the body; any rule that is more urgent than another must have its predicate computed before the other; and any rule that executes before another must have its body before the other.  But there are also edges for paths through state: a rule that writes to a Wire must have its body execute before the predicate or body of another rule that reads the Wire (in its predicate or body).  Having already reported execution and urgency cycles, if there any loops in this combined graph, BSC reports it as a more complicated cycle that needs to be resolved.  BSC also uses these graphs to warn about underspecified orders (where BSC has to pick a direction, and then insert that info into the graph); and there are attempts at "biasing" the flattened order to do things like put all methods before rules, say.  At the end, the scheduling relationships/annotations for the methods are computed by looking the graph (usually by looking to see if there is path through the graph from the nodes for method1 to the nodes for  [... cut ...] ody/action of a single rule).  An interesting case that is handled is when two methods have a rule that must execution between them; in that case, the methods are not parallel composable (even if they normally would be), because the rule in between prevents them from coming together; that "rule between" has to be taken into account when creating the annotations for the interface.

Anyway, that's some things off the top of my head.  Maybe @rsnikhil or others can chime in if I've missed something.

ThePerfectComputer, 2022-10-23:
This is a good response and give me a lot to get started with. I didn't know about
[BSVLang/Main](https://github.com/BSVLang/Main).

ThePerfectComputer, 2022-10-27:
I noticed there is ``bsc2bsv`` and a ``bsv2bsc`` - but these targets don't seem to be built by default. Is this intentional?

ThePerfectComputer, 2022-10-27:
nvm, make install-utils seems to handle that