B-Lang-org/bsc #517: deriving Bitwise for enum

pwang7, 2022-11-21:
Hi there, how to implement `Bitwise` for `enum`?

I have some kind of flags implemented as `enum`, and I'd like it support bitwise operations among flags, such `A | B`, `C | B`:
```Bluespec
typedef enum {
  A = 1,
  B = 2,
  C = 4,
  D = 8
} Flags deriving(Bits, Bitwise, Eq);
```
But deriving `Bitwise` for `enum` is not allowed in BSV.

krame505, 2022-11-21:
Deriving isn't supported for the `Bitwise` type class, you would need to write an instance manually.  You can define it in terms of the `Bit` instance for `Flags`, e.g.
```
instance Bitwise Flags where
  (&) x y = unpack (pack x & pack y)
  ...
```
using the classic syntax, I'm not familiar with the BSV syntax but it should be logically the same.

pwang7, 2022-11-25:
> Deriving isn't supported for the `Bitwise` type class, you would need to write an instance manually. You can define it in terms of the `Bit` instance for `Flags`, e.g.
>
> ```
> instance Bitwise Flags [...]

Thanks!

closed 2022-12-05 by quark17