B-Lang-org/bsc #518: repeatFSMStmt unexpected behaviour

timksf, 2022-11-23:
Hello,

I have encountered some unexpected behaviour using a repeat statement, similar to #280 . In this particular case, the repeat statement does not appear to be equivalent to a while loop, as it should be according to the reference guide. I have reduced the issue to the following:

```bluespec
import StmtFSM :: *;
import FIFO :: *;
import GetPut :: *;
import ClientServer :: *;

(* synthesize *)
module mkTest(Server#(int, int));
    Integer wait_cycles = 4;

    FIFO#(int) fifo_in <- mkFIFO;
    FIFO#(int) fifo_out <- mkSizedFIFO(10);
    Reg#(int) cnt <- mkReg(fromInteger(wait_cycles));
    Reg#(Maybe#(int)) bf <- mkReg(tagged Invalid);

    rule rstart (bf matches tagged Invalid);
        cnt <= fromInteger(wait_cycles);
        bf <= tagged Valid fifo_in.first;
        fifo_in.deq;
    endrule

    rule rproc (cnt > 0 && isValid(bf));
        cnt <= cnt - 1;
    endrule

    rule rfin (cnt == 0 &&& bf matches tagged Valid .v);
        fifo_out.enq(v);
        bf <= tagged Invalid;
    endrule
    interface request = toPut(fifo_in);
    interface response = toGet(fifo_out);
endmodule

module mkTestbench();

    Reg#(int) cnt <- mkReg(0);
    Reg#(int) cnt1 <- mkReg(0);

    let dut <- mkTest;

    Stmt s_repeat = seq
        repeat(10) action
            $display("[%04d]Wrote: %02d", $time, cnt);
            dut.request.put(cnt);
            cnt <= cnt + 1;
        endaction
        seq
        while(cnt1 < 10) action
            let x <- dut.response.get;
            $display("[%04d]Read:  %02d", $time, x);
            cnt1 <= cnt1 + 1;
        endaction
        endseq
    endseq;

    Stmt s_while = seq
        while(cnt < 10) seq
            action
            $display("[%04d]Wrote: %02d", $time, cnt);
            dut.request.put(cnt);
            endaction
            cnt <= cnt + 1;
        endseq
        seq
        while(cnt1 < 10) action
            let x <- dut.response.get;
            $display("[%04d]Read:  %02d", $time, x);
            cnt1 <= cnt1 + 1;
        endaction
        endseq
    endseq;

    mkAutoFSM(s_repeat);
    //mkAutoFSM(s_while);

endmodule
```

The counter generated for the repeat statement should only be incremented once, when the loop body fires. In the waveforms you can see however, at 50us the counter is incremented even though the loop body did not fire. The reason appears to be the CAN/WILL_FIRE for the counter update action, which is held one cycle longer than the CAN/WILL_FIRE of the loop body.
[image]
I've also attached the schedules generated for the while and the repeat variant.

Regards, Tim
[TestbenchRepeatSched.txt](https://github.com/B-Lang-org/bsc/files/10077591/TestbenchRepeatSched.txt)
[TestbenchWhileSched.txt](https://github.com/B-Lang-org/bsc/files/10077592/TestbenchWhileSched.txt)

quark17, 2022-12-05:
I want to make sure that I understand the behavior difference you're seeing.  If I simulate the testbench with `while`, I see this output and the simulation ends:
```
[0015]Wrote: 00
[0035]Wrote: 01
[0055]Wrote: 02
[0095]Wrote: 03
[0155]Wrote: 04
[0215]Wrote: 05
[0275]Wrote: 06
[0335]Wrote: 07
[0395]Wrote: 08
[0455]Wrote: 09
[0475]Read:  00
[0485]Read:  01
[0495]Read:  02
[0505]Read:  03
[0515]Read:  04
[0525]Read:  05
[0535]Read:  06
[0545]Read:  07
[0565]Read:  08
[0625]Read:  09
```
If I simulate the testbench with `repeat`, then I see this and the simulation hangs:
```
[0015]Wrote: 00
[0025]Wrote: 01
[0035]Wrote: 02
[0095]Wrote: 03
[0155]Wrote: 04
[0215]Wrote: 05
[0275]Wrote: 06
[0335]Wrote: 07
[0395]Wrote: 08
[0415]Read:  00
[0425]Read:  01
[0435]Read:  02
[0445]Read:  03
[0455]Read:  04
[0465]Read:  05
[0475]Read:  06
[0505]Read:  07
[0565]Read:  08
```
So the issue is that the `repeat` statement is only executing the body 9 times even though the argument given was 10.  And you've identified that the rule for incrementing the counter executes a tenth time, but the rule for executing the body does not execute that last time.  Have I got that right?  That certainly seems like a bug -- sorry about that!

timksf, 2022-12-10:
Yes, exactly that. No worries, using a while loop is sufficiently convenient anyways.

quark17, 2023-09-29:
I discovered that this issue exists in Bluespec Inc's proprietary bug database as issue 1790.  And that the example given is in the testsuite, at `bsc.bsv_examples/SHA256/KenSha2.bsv`.  There is a module `bug` in that file which exhibits the issue (and is not what the testsuite uses).  The following commands will illustrate the issue, by having only two lines of output rather than the three expected:
```
bsc -sim -g bug KenSha2.bsv
bsc -sim -e bug
./a.out
```