B-Lang-org/bsc #522: bsc insert delays for system tasks.

quentinschibler, 2022-12-14:
When using system tasks like $display, bsc insert a couple of #0 delays. I'm not sure if they are useful for event based simulator, but when using verilator it generates errors because delays are not supported. I haven't seen any option to disable them, maybe they should be converted to BSV_ASSIGNMENT_DELAY ?

quark17, 2022-12-16:
The Verilator documentation explains how to disable some errors and warnings, [here](https://verilator.org/guide/latest/warnings.html).  In this case, you can provide the following statement in a config file provided on the command-line:
```
lint_off -rule STMTDLY
```
An example configuration file is in the BSC repo [here](https://github.com/B-Lang-org/bsc/blob/main/src/Verilator/verilator_config.vlt).  This file is used by BSC when it is asked to link a simulation using Verilator:
```
bsc -e mkTopModule -verilog -vsim verilator
```
This support in BSC was added not long ago, but Verilator has been changing fast, and some of the support can already be simplified.  For example, it used to be that you had to provide a C++ `main` to drive the Verilator-generated model.  Many users just want to create a simulation from their design in the same way they would with any other Verilog simulator, where your testbench is in Verilog (or SystemVerilog) and no C++ driver is needed.  Verilator is now trying to support that use model.  The flag `--binay` can be provided on the command-line, to tell Verilator to behave in this way.

This is part of a larger effort for Verilator to support more of the non-synthesizable subset of Verilog.  That includes support for timing delays!  The recently-released Verilator v5 does have support for delay statements.  I believe the flag for that is `--timing`.  However, it may have an impact on performance -- I don't know enough about how it's implemented and I haven't experimented with it yet.

But you shouldn't need to provide `--timing` if you just want Verilator to ignore the `#0` delays in the system task always blocks.  Turning off Verilator's `STMTDLY` error should be sufficient.

The reason for `#0` is because you might have Verilog like this:
```
assign v_plus_one = v + 1;

always@(negedge CLK)
begin
  f = $fopen("filename", "r");
  v = $fgetc(f); // statement A
  #0;
  $display(v_plus_one); // statement B
end
```
Here, we want the simulator to start executing the always block by opening the file and getting a value from it.  The value is assigned to `v` and then we want to display a signal `v_plus_one` which is defined based on `v`.  For `v_plus_one` to update. the simulator must execute the continuous assignment between execution of those statements A and B.  My understanding is that the delay statement is needed, to tell the simulator that it can suspend execution of the current always block, go and execute any other pending work (such as other always blocks or continuous assignments that have triggered) and then return to this always block.  It has been a while since I've had to think about the execution model of Verilog, so I can't recall if the Verilog spec specifically says that `v_plus_one` will not update unless a delay is inserted, or if it's just that some simulators will choose not to.  I'm tempted to say that it's part of the spec, which makes me wonder what behavior Verilator gives when told to ignore the delay.  Will Verilator display a stale value?  It'll be worth making a small example and testing.

This situation only comes up when executing system tasks that return a value (such as `fgetc`), whose value is then fed as input to another system task.  I think most designs don't do this, so it probably won't come up for you, and telling Verilator to ignore the delay will be fine.  BSC blindly inserts the delay everywhere, so that it's there when needed; we could instead have BSC do some extra work to analyse which situations need the delay and to only insert the delay when needed -- so that most generated Verilog would not have the delay.

FYI, the `BSV_ASSIGNMENT_DELAY` macro would not be appropriate here.  That macro is for register updates, so that when you simulate and generate a VCD, the waveform will show a small delay between the clock edge and when the registers update to the new value.  Some people (or companies) use this debug methodology, and the macro is there to support them.  The macro can be set to a number, to introduce a state update delay of that number of steps.  (The delay in always blocks for system tasks is not about register updates and would not use this macro.)

quark17, 2022-12-16:
Here I've shown a simple case of one continuous assignment.  But there could be more complicated scenarios, involving always blocks in other modules, which are also executing system tasks.  If the delay isn't needed for this simple example, it's possible that the delay is needed for a more complicated situation like that.  I apologize that I've forgotten the history.  There may be an example in the testsuite that illustrates the need (and could be found by removing the delay-generation from BSC and running the testsuite to see if anything fails).

quark17, 2023-01-17:
I found a test case for "logic between system tasks" in `testsuite/bsc.verilog/tasks/LogicBetweenTasks.bsv`.  The test calls `$stime` to get a value, does not operations on the result (inverting and AND'ing with a register), and then calls `$display` on the result.  When this test was written, the assignments for the intervening logic were probably generated as global assignments; but there was an improvement to BSC (in the last five years?) to inline the assignments into the always block for the tasks.  In that case, the delay (`#0`) is not needed (to yield execution from the always block and allow the assignment statements to execute), because the assignments are in the always block.

Given that inlining, it is even rarer that a design would need the zero delay for correct operation.

However, I have constructed such an example.  It is based on `LogicBetweenTasks.bsv`, but the intervening logic is in a separately synthesized submodule, so it cannot be inlined.  The file is 
[Test.bsv](https://github.com/B-Lang-org/bsc/files/10430329/Test.bsv.txt) and [Makefile](https://github.com/B-Lang-org/bsc/files/10430328/Makefile.txt) demonstrates the difference in behavior with and without the zero delay.

Interestingly, Verilator ignores delay statements, and so it gets the behavior wrong!  (Starting with version 5, Verilator requires a flag to explicitly specify whether to ignore delays or to handle them.  But it currently does not expect to find zero delays, and crashes!)  Another further interesting fact is that Icarus (iverilog) and Verilator report different behavior for the version with no hash statement!  That is also demonstrated in the Makefile (if you have both simulators installed).

quark17, 2023-01-29:
With version 5 of Verilator, the user is now required to supply either `--timing` or `--no-timing`, for designs that contain `#`.  With earlier versions, Verilator would just ignore `#` in the source and implement the design as if it wasn't there.  This behavior can be achieved by providing `--no-timing`.  Interestingly, with `--timing`, Verilator will still reject `#0` as unsupported.  An issue has just been opened on the Verilator repo (issue 3910) to support it (presumably with `--timing`).

BSC's uses of `#0` should not require the dynamic runtime overhead that comes with `--timing` (or that I assume comes with it).  The design inside BSC has been statically scheduled, and BSC is just limited by the Verilog language in how it can represent the design.  Hopefully, through static analysis, Verilator could reconstruct the sequence of assignments and not need dynamic looping and event queues.  But I don't know if this is possible.  At the very least, hopefully Verilator will support `#0` soon with `--timing`.

closed 2023-02-14 by quentinschibler