B-Lang-org/bsc #543: BSC cannot synthesize polymorphic modules

quentinschibler, 2023-02-09:
BSC is not able to synthesize a Verilog module from polymorphic modules in Bluespec. I think it is doable if you pass the bit-width of your type as a Verilog parameter. 
The only way to do something similar is to instantiate a wrapper with the specific size you want, and (* synthesize *) that. It doesn't work if you have one level of indirection, let's say an adder inside an ALU. If you use an adder wrapper with a word width of 32 bits, then your ALU has to be 32 bits wide, or you need to implement it twice.
This is an important issue for two reasons. First when you look at the RTL view in your CAD tool, if every module is flattened out, you have zero chance of understanding what is going on with your design. Secondly, at least with Quartus, the synthesis results of a flattened designs are of a poor quality compared to a modular design, for my specific use case I can get a decrease in performance of 35%.

Aaron-Hartwig, 2023-02-14:
I'm also curious about this limitation and why `bsc` does not created parameterized Verilog.

quark17, 2023-02-15:
Yes, this is a feature that would be good to have.  It's worth noting that polymorphism can be used in a spectrum of ways; there are probably some basic uses that can be easily represented in Verilog, but there are many uses that would be difficult to represent and some that are probably impossible.  If we start by supporting some basic cases, BSC would need to gracefully report to the user when their code is still not synthesizable.  Also, BSC's elaboration was not designed with this in mind, so some effort may be involved in implementing it.  I'm available to help anyone who is interested in taking this on, but I don't have the resources to do it.

Generating a single parameterized Verilog representation is not currently possible, but in some cases people are just concerned about having a Verilog module boundary, and for that, several workarounds exist.  As you noted, one can manually wrap a polymorphic module, to synthesize multiple instances of it:
```
// A module that is polymorphic in the size n
module mkMod #(Ifc#(n)); ... endmodule

// A module for the specific size 8
(* synthesize *)
module mkMod_8 #(Ifc#(8));
  (* hide *)
  let __i <- mkMod;
  return __i;
endmodule

// A module for the specific size 16
(* synthesize *)
module mkMod_16 #(Ifc#(8));
  (* hide *)
  let __i <- mkMod;
  return __i;
endmodule
```
(In this case, I've used the `hide` attribute and named the instance beginning with underscores, which tell BSC not to include the name or hierarchy in various places, such as in the generated Verilog names or in messages or info that Bluetcl reports about the design.  I forget offhand if both of these forms are needed.)

The above code requires you to instantiate the specific name where you use it:
```
module mkTop(...);
  // The context requires size 8
  let i <- mkMod_8;
  ...
endmodule
```
That's inconvenient and fragile.  But fortunately there's a workaround for this too.  You can define a typeclass that overloads the name `mkMod`:
```
typeclass MkMod#(type t);
  module mkMod(Ifc#(t));
endtypeclass

instance MkMod#(8);
  module mkMod(Ifc#(8));
    (* hide *)
    let __i <- mkModCore;
    return __i;
  endmodule
endinstance

instance MkMod#(16);
  module mkMod(Ifc#(16));
    ...
  endmodule
endinstance
```
With the above, you can then use a single name wherever you want to instantiate that module:
```
module mkTop(...);
  let i <- mkMod;
  ...
endmodule
```
The specific instance that is used will depend on the size expected by the context where it's called.  If an instance doesn't exist for the size, you'll get a type-check error.  You provide an alternate behavior (such as a better error message or default to a non-separately-synthesized implementation) by defining a catch-all instance:
```
instance MkMod#(n);
  module mkMod(Reg#(Bit#(n)));
    let	__i <- mkMod;
    String msg = strConcat("No instance of 'mkMod' for size ", printType(typeOf(__i)));
    errorM(msg);
  endmodule
endinstance
```
[... 61 lines cut ...]
```
module mkMod (Ifc#(sz));
  ...
  SubIfc#(sz) s;
  if (valueOf(sz) == 8)
    s <- mkMod8;
  else if (valueOf(sz) == 16)
    s <- mkMod16;
  else
    errorM("Unsupported size");
  ...
endmodule
```
In this case, maybe the design could be represented in Verilog that instantiates both and muxes the appropriate one?  But later stages of BSC would need to support both possibilities.  For example, what if the two modules being instantiated have different scheduling relationships?  The scheduling stage would need to handle both possibilities.  Of course, we can start by not supporting synthesis of designs with conditional instantiation; or maybe we allow it if the two instances turn out to have the exact same scheduling info.

There are also designs with loops:
```
for (Integer i=0; i < valueOf(sz); i = i + 1)
  ...
```
It may be possible to implement this in Verilog using a loop.  Or maybe the logic that results from the loop can be rephrased in a way that doesn't use a loop.  Either way, BSC currently works by fully unrolling loops, so it would not (currently) be able to evaluate this code without knowing the exact value of `sz`.  We would like to improve BSC to not unroll loops unnecessarily, though, so that would separately be a good thing to do.  (The higher level that we can output the Verilog or the Bluesim C++, the better.)

A design written with a loop like the above could possibly be re-coded by the user to express it in a way that doesn't use a loop.  If BSC supported synthesis of only some basic polymorphism, it's possible that users may need to rephrase their code, to get it into a form that BSC can compile.

Anyway, those are just some of the considerations of why implementing this feature may be hard.  But there's probably some basic cases that we can start by implementing, if someone wanted to take this on.

quark17, 2023-02-15:
> This is an important issue for two reasons. First when you look at the RTL view in your CAD tool, if every module is flattened out, you have zero chance of understanding what is going on with your [...]

It sounds like maybe the `typedef 8 N` approach (that I mentioned) would satisfy your needs?  This is used in projects like [Flute](https://github.com/bluespec/Flute).  There, the specific parameters can even be controlled from a Makefile.  For example, [here](https://github.com/bluespec/Flute/blob/14ccd91b39c1c1ce55e548492c45175b8dcb1e24/src_Core/ISA/ISA_Decls.bsv#L40) the choice of defining `RV32` or `RV64` (via `-D` on the BSC command-line) controls the data width size (`XLEN`).  Modules can be written using this variable name (or conditionally instantiating state and logic based on it) and still be synthesizable.

I'll also mention that BSC does keep track of the inlined hierarchy and that Bluetcl provides ways of seeing that hierarchy and getting access to any signals might still be visible, and the [BDW](https://github.com/b-lang-org/bdw) GUI allows viewing that in a graphical browser (and clicking on instances to add the related signals to an attached waveform viewer, like gtkwave).  If your RTL tools have plugins or hooks (like gtkwave does), it may be possible to create tools that provide source-level views (or closer to it) via Bluetcl.  (Also, the BDW has a code that hooks into Gtkwave to display structs/unions/enums with their source-code names and fields, rather than as raw bits.)

quark17, 2023-02-15:
In Flute, the [register file module](https://github.com/bluespec/Flute/blob/master/src_Core/RegFiles/CSR_RegFile_MSU.bsv) is even written to have different methods in its interface depending on what options are defined (and there's a lot of conditional code in the module), while still having a `synthesize` attribute.

quentinschibler, 2023-02-20:
Thanks for the explanation, the typedefs cover most of my need, but in a few places I still need polymorphism. I am probably going to write the wrappers for them as you described. I sadly don't have time either to try to implement it, I'm keeping the issue open in case someone wish to try.

quentinschibler, 2023-03-03:
I tried using the typeclass method explained above, but I don't think it solves the issue. If you have two modules, `mkChildCore(Itf#(width))` and `mkParentCore(Itf#(width))`, both polymorphic, with a bunch of `mkChild` instances returning a `mkChildCore`, bsc will fail with "no instance of the form MkChild#(width)" if you try to instanciate a child in mkParentCore

quark17, 2023-03-03:
In the future, it would help to show an example, or at least some code snippets.  I suspect you might need a proviso on `mkParentCore`.  If you use a polymorphic typeclass method (in this case a module) inside a module or function that is itself polymorphic, and the context of the use is not a fixed type but is polymorphic in the parent's type variables, then you need a proviso saying that an instance for the child will exist when the parent's polymorphism is resolved.  Here's an example:

First, declare a child:
```
typeclass MkChild#(numeric type n);
  module mkChild(Reg#(Bit#(n)));
endtypeclass

module mkChildCore (Reg#(Bit#(n)));
  Reg#(Bit#(n)) rg <- mkRegU;
  return rg;
endmodule

(* synthesize *)
module mkChild8 (Reg#(Bit#(8)));
  (* hide *)
  let __i <- mkChildCore;
  return __i;
endmodule

instance MkChild#(8);
  function mkChild = mkChild8;
endinstance
```
Then, declare the parent:
```
typeclass MkParent#(numeric type n);
  module mkParent(Reg#(Bit#(n)));
endtypeclass

module mkParentCore (Reg#(Bit#(n)))
provisos (MkChild#(n));
  Reg#(Bit#(n)) ch <- mkChild;
  return ch;
endmodule

(* synthesize *)
module mkParent8 (Reg#(Bit#(8)));
  (* hide *)
  let __i <- mkParentCore;
  return __i;
endmodule

instance MkParent#(8);
  function mkParent = mkParent8;
endinstance
```
Here, in `mkParentCore`, the context for the instantiation of `mkChild` is not concrete, it's a variable.  So a proviso is needed.  When BSC typechecks `mkParentCore`, it doesn't know which instance of `MkChild` is need; the proviso delays the decision until the typechecking of the place where `mkParentCore` is instantiated.  The proviso says that the context for `mkParentCore` will be responsible for determining which instance to use.

If you have a fall-though instance `MkChild#(a)`, then you have overlapping instances, because when `MkChild#(8)` is applicable, `MkChild#(a)` is also applicable.  BSC allows overlapping instances, and deals with it by choosing the instance that most concretely matches.  So if both `8` and `a` are applicable, BSC will choose `MkChild#(8)`.  But inside `mkParentCore`, when the concrete type is not yet known, the best that BSC can do is to pick `MkChild#(a)`.  If you put a proviso, then BSC doesn't have to pick; but if you leave the proviso off, then BSC is forced to resolve it at that point, with no information, and has to choose `MkChild#(a)`.  If you hadn't declared a fall-through instance, BSC would have given you an error that no instance was found, and that error message would have included a suggestion that you might want to add the proviso.  (So by defining the fall-through, you miss out on the more helpful message, unfortunately.)

Anyway, for the example I've shown above, we can write a test to see that it works:
```
(* synthesize *)
module mkTest ();
  Reg#(Bool) done <- mkReg(False);
  Reg#(Bit#(8)) p <- mkParent;

  rule r1 (!done);
    p <= 17;
    done <= True;
  endrule

  rule r2 (done);
    $display("p = %d", p);
    $finish(0);
  endrule
endmodule
```
And I do see it print `p = 17`.

If the relationship between the parameter of `mkChild` and `mkParent` is not equality, then you might need different provisos.  For example:
```
module mkParentCore (Reg#(Bit#(n)))
provisos (Div#(n, 2, halfn),
          Add#(halfn, halfn, n),
          MkChild#(halfn));
  Reg#(Bit#(halfn)) ch1 <- mkChild;
  Reg#(Bit#(halfn)) ch2 <- mkChild;

  method Action _write(Bit#(n) x);
    ch1 <= truncate(x);
    ch2 <= truncateLSB(x);
  endmethod
  method Bit#(n) _read() = { ch2, ch1 };
endmodule
```

quentinschibler, 2023-03-04:
Thanks a lot for your detailed answer, I didn't know you could put a provisos to delay choosing the right instance. I'll definitely include code sample in the future.