B-Lang-org/bsc #545: bsc does not enforce the correct use of always_enabled/ready methods

quentinschibler, 2023-02-15:
Using always_enabled methods within a rule that have fire conditions, or using a method with implicit conditions inside an always_ready methods don't trigger any warning or error. It is explicitly said in the reference guide that "The methods will be executed on every clock cycle and the compiler verifies that
the caller does this.", which is not the case.
bsc version is 2022.01, I am updating to 2023.01 to see if it fixes the issue, but I haven't seen anything related to that in the changelog.

quark17, 2023-02-15:
Is the submodule separately synthesized?  BSC doesn't check attributes on inlined modules (and doesn't warn that they are being ignored).  This is a known issue -- in Bluespec Inc's internal bug database, it is issue 953, from 2005.  I think it's a feature that people have just lived with, but it could perhaps be better documented.  (Or a warning added in some way.)

If the submodule is synthesized, and you're still seeing a problem, can you provide an example?

quentinschibler, 2023-02-20:
You're correct, putting a synthesize boundary on the module fixed it. A warning would be great, but documenting it would be fine. By the way, do we have access to the internal bug database ? I assume internal probably means no, but asking anyway.

quark17, 2023-03-10:
> do we have access to the internal bug database ?

No.  The entries in the database may contain sensitive information, such as private code that customers submitted, or even just the names of customers.  We do want to migrate the entries over to GitHub, by rewriting the code or comments to leave out any sensitive information.  But it is a manual process, that only employees at Bluespec can do.  I started moving a few, but I'm not sure when I or others will have time to move them all over.