B-Lang-org/bsc #552: Conditional expressions with pattern matching does not compile

pbing, 2023-04-23:
I cannot compile conditional expressions with pattern matching.

This compiles:
```BSV
// 9.6 Conditional statements
(* noinline *)
function Bool func1(Maybe#(int) x);
   if (x matches tagged Valid .v &&& v >= 0)
      return True;
   else
      return False;
endfunction
```

But this does not:
```BSV
// 10.2 Conditional expressions (compile error)
(* noinline *)
function Bool func2(Maybe#(int) x);
   return (x matches tagged Valid .v &&& v >= 0) ? True : False;
endfunction
```
```shell
% bsc -verilog -u ./src/Testcase.bsv
checking package dependencies
Error: "./src/Testcase.bsv", line 13, column 48: (P0005)
  Unexpected `)'; expected `&&&' or `?'
```

quark17, 2023-04-23:
Both of the following are accepted syntax by BSC:

return x matches tagged Valid .v &&& v >= 0 ? True : False;

return (x matches tagged Valid .v &&& v >= 0 ? True : False);

What you have written puts parentheses around the "matches" clause:

return (x matches tagged Valid .v &&& v >= 0) ? True : False;

That is not valid syntax.  BSC allows you to put extra parentheses around
an expression, if you want, but the "matches" clause is not an expression.
That is, you CANNOT write the following:

Bool y = matches tagged Valid .v &&& v >= 0;

BSC will complain that it expects an expression after the equals, and a
matches clause is not an expression.

If you look at the grammar in the BSC language guide, in Section 9.6
(Conditional statements), you will see that the condition in the
parentheses of an if-statement can be EITHER an expression or a matches
clause:

condPredicate ::= exprOrCondPattern { &&& exprOrCondPattern }

exprOrCondPattern ::= expression | expression matches pattern

So when you write this:

if (e matches p)

It is not being parsed as this:

if (expression)

Instead, the "matches" clause is being specially handled as an option for
that syntax.

The "matches" clause (and the "&&&" operator) is specially handled for
method conditions (Section 5.5 Interface definition), for rule conditions
(Section 5.6 Rules in module definitions), for if-statements (Section 9.6
Conditional statements), and the ternary conditional operator (Section 10.2
Conditional expressions).  The grammar does not include "matches" as one of
the forms for "expression".  And thus, the syntax does not allow you to add
extra parentheses around the clause.

In theory, the syntax for those four situations (rules, methods, and
conditionals) could be changed to allow extra parentheses.  But I don't
know off-hand if maybe that's ambiguous or something, that would make it
difficult to parse.

Julie

On Sun, Apr 23, 2023 at 9:11 PM pbing ***@***.***> wrote:

> I cannot compile conditional expressions with pattern matching.
>
> This compiles:
>
> // 9.6 Conditional statements(* noinline *)function Bool func1(Maybe#(int) x);
>    if (x matches tagged Valid .v &&& v >= 0)
>       return True;
>    else
>       return False;endfunction
>
> But this does not:
>
> // 10.2 Conditional expressions (compile error)(* noinline *)function Bool func2(Maybe#(int) x);
>    return (x matches tagged Valid .v &&& v >= 0) ? True : False;endfunction
>
> % bsc -verilog -u ./src/Testcase.bsv
> checking package dependencies
> Error: "./src/Testcase.bsv", line 13, column 48: (P0005)
>   Unexpected `)'; expected `&&&' or `?'
>
>

quark17, 2023-04-23:
Both of the following are accepted syntax by BSC:
```
return x matches tagged Valid .v &&& v >= 0 ? True : False;

return (x matches tagged Valid .v &&& v >= 0 ? True : False);
```
What you have written puts parentheses around the "matches" clause:
```
return (x matches tagged Valid .v &&& v >= 0) ? True : False;
```
That is not valid syntax.  BSC allows you to put extra parentheses around an expression, if you want, but the "matches" clause is not an expression. That is, you CANNOT write the following:
```
Bool y = matches tagged Valid .v &&& v >= 0;
```
BSC will complain that it expects an expression after the equals, and a matches clause is not an expression.

If you look at the grammar in the BSC language guide, in Section 9.6 (Conditional statements), you will see that the condition in the parentheses of an if-statement can be EITHER an expression or a matches clause:
```
condPredicate ::= exprOrCondPattern { &&& exprOrCondPattern }

exprOrCondPattern ::= expression | expression matches pattern
```
So when you write this:
```
if (e matches p)
```
It is not being parsed as this:
```
if (expression)
```
Instead, the "matches" clause is being specially handled as an option for that syntax.

The "matches" clause (and the "&&&" operator) is specially handled for method conditions (Section 5.5 Interface definition), for rule conditions (Section 5.6 Rules in module definitions), for if-statements (Section 9.6 Conditional statements), and the ternary conditional operator (Section 10.2 Conditional expressions).  The grammar does not include "matches" as one of the forms for "expression".  And thus, the syntax does not allow you to add extra parentheses around the clause.

In theory, the syntax for those four situations (rules, methods, and conditionals) could be changed to allow extra parentheses.  But I don't know off-hand if maybe that's ambiguous or something, that would make it difficult to parse.

pbing, 2023-04-24:
Thank you for the explanation!

closed 2023-04-24 by pbing