B-Lang-org/bsc #553: Inout interface different clock domains

mounakrishna99, 2023-05-04:
Hi,

I am taking an inout port of a verilog module and exposing it to the top level module. The verilog module is at a different clock domain compared to the top level module and I am not able to connect them.  
How to change the clock domain of the Inout ports declared in the top level to the clock domain of the verilog module?

mounakrishna99, 2023-05-05:
Found the way. 
Of the two ways in mentioning Inout ports, one as method parameters and other as subinterfaces. Writing the Inout ports as method parameters I am facing the issue of different clock domains. Changing them as subinterfaces, it worked :)

closed 2023-05-05 by mounakrishna99

quark17, 2023-05-05:
The way your originally wrote it might work too.  But without seeing the code you wrote, I can't say what you would need to do differently.  In the future, it would help to show some actual code, not just words.  (And also show the error message.)

Putting the Inout in the interface is probably the better approach; so if you've got that working, that's great!  For the other way, you may have needed to specify the clock somewhere.  Be aware that you can add attributes or clauses in various places to specify the associated clock: on module ports/arguments, on submodule instantiations (including mkConnection), and on import-BVI ports and interfaces.