B-Lang-org/bsc #556: Don't want clock to be in module boundary

mounakrishna99, 2023-05-24:
Hi,
I have a design in which some internal blocks are clocked by a different clock and it's IO pins being exposed to the top module. Like let's say I have a Soc with core, uart, spi, adc, pll where the adc is being clocked by a different clock and it's IO pins exposed to top module.

When I generate verilog for the Soc, the compiler gives error stating the adc_clock is not available in module boundary. But I don't want those pins at the top module ports as I don't need any clock for those analog pins. I tried making those methods (pins) as clocked_by(no_clock) which removed the "Clock not availble in module boundary" error, but they were creating unnecessary EN_.. and RDY_.. pins at the top module. Putting (always_enabled, always_ready) for those pins didn't work as the compiler throwed error saying the condition to execute that method is always False as no_clock and can't be always enabled.

Is there any compile option to ignore that check?
And is there any forum for bluespec discussions? I couldn't find any.

Thanks
Mouna Krishna

rsnikhil, 2023-05-24:
>    And is there any forum for bluespec discussions? I couldn't find any.

    https://groups.io/
        b-lang-announce
        b-lang-discuss

mounakrishna99, 2023-05-24:
Thanks, I found a discussion regarding the issue I am facing. I will try that out.
https://groups.io/g/b-lang-discuss/topic/81494353#39

closed 2023-05-24 by mounakrishna99