B-Lang-org/bsc #561: Test fails with IVerilog v12

Vekhir, 2023-06-02:
#### Description
When running `make check` with IVerilog v12 installed, the tests `bsc.interra/operators/Arith/arith.exp` and `bsc.verilog/positivereset/ClockDividers/ClockDividers.exp` fail. This does not occur when running with IVerilog v11.\
This only affects the tests, but is annoying nonetheless. In any case, the tests should obviously always pass.

Do you have ideas where the error could be coming from? Any pointers where I could start to find the cause (either in the tests or IVerilog) are appreciated.

Running `make smoke` is also unaffected since the failing tests are not included there.

#### Errors
##### bsc.interra/operators/Arith/arith.exp
FAIL: module `mkTestbench` in `Testbench.bsv` should compile to Verilog
##### bsc.verilog/positivereset/ClockDividers/ClockDividers.exp:
FAIL: `sysClockDiv.v ClockDiv.v` should link to executable `sysClockDiv`

#### Appendix (Full test log)
[errorlog.txt](https://github.com/B-Lang-org/bsc/files/11634904/errorlog.txt)

quark17, 2023-06-10:
The testsuite only prints very high-level information to the terminal (what you recorded in `errlog.txt`).  It says when it's entering a test subdirectory:
```
Running /path/bsc/testsuite/bsc.interra/operators/Arith/arith.exp ...
```
and the `.exp` file may call procedures like `note` to print informational messages:
```
NOTE: Random seed = 3427
```
But otherwise, the testsuite won't print any message unless a test fails (`FAIL:`) or unexpectedly passes (`XPASS:`) -- the later usually indicating that a feature has been fixed and the test for it is now passing, so the expected behavior just needs to be updated).

There is a lot more information that is recorded in files, which you can look at.  It's worth noting that a test is usually multiple steps (such as compile, link, simulate, compare the simulation output to the expected), and if the compile step fails, then you'll also see failures for the other steps:
```
FAIL: module `mkTestbench' in `Testbench.bsv' should compile to Verilog
FAIL: `mkTestbench.v mkDut.v' should link to executable `mkTestbench'
FAIL: Verilog simulation `mkTestbench' should execute: child process exited abnormally
FAIL: `mkTestbench.v.out' differs from `arith.out.expected'
```
So you'll want to look at the first failure message for a given test, and can ignore the rest (that have the same name).  Also, we test both Verilog and Bluesim backends.  Often, the compile step is done once, for Verilog, and the generated `.ba` files are reused for Bluesim linking, so that the compile step doesn't have to be run twice.  So if the compile step fails, you might see both Verilog and Bluesim linking failures that follow.  (Some `.exp` files might separately test Bluesim and Verilog; so it depends on how the test is written.)  We also test both Verilog and Bluesim a second time, with VCD dumping turned on; so the simulation step may run twice.  All of that is just to explain why you also see the following additional failures (after the above failures):
```
FAIL: Verilog simulation `mkTestbench' should execute with VCD dump: child process exited abnormally
FAIL: `mkTestbench mkDut' should link to executable `mkTestbench'
FAIL: Bluesim simulation `mkTestbench' should execute: child process exited abnormally
FAIL: `mkTestbench.c.out' differs from `arith.out.expected'
FAIL: Bluesim simulation `mkTestbench' should execute: child process exited abnormally
```
All of these failures in the `bsc.interra/operators/Arith/` directory are due to the initial BSC compile failure.

The output that you see on the terminal is recorded in a file `testrun.sum` ("sum" for summary).  A more verbose log is written to the file `testrun.log`.  That file outputs the commands that are being executed, so you can see what specific command failed:
```
Executing (bsc_compile_verilog): /Users/quark/Work/git-repos/blang/bsc/inst/bin/bsc  -elab -no-show-timestamps -no-show-version -u -verilog -g mkTestbench Testbench.bsv >& Testbench.bsv.bsc-vcomp-out
FAIL: module `mkTestbench' in `Testbench.bsv' should compile to Verilog
```
This is telling you that the test procedure `bsc_compile_verilog` is being run, and the command that it is running, including showing the file that stdout and stderr are being piped to.  The output of every command is recorded to a file, which you can examine afterwards.  The `testrun.log` file will tell you the name, but once you learn the pattern, you'll know what to look for without having to open the `testrun.log` file.  The files have a suffix, of the form `<description>-out`, and so you can list the files ending in `out`, to see what's generated.  Sometimes there are further extensions -- for example, before comparing the output to an expected file, the testsuite might filter the file, and that goes to `<description>-out.filtered`.  So listing all the `*out*` files might show you more.  Anyway, the output of the BSC compile step is `<Foo.bsv>.bsc-vcomp-out` for Verilog or `<Foo.bsv>.bsc-ccomp-out` for Bluesim, or even just `<Foo.bsv>.bsc-out` for compilation with no backend flags.  The linking step adds the suffix to the top module name, so for example, `<mkFoo>.bsc-vcomp-out`.  And the simulation output has the form `<mkFoo>.out` if Bluesim and Verilog are expecting the same output, otherwise `<mkFoo>.c.out` and `<mkFoo>.v.out`.  If the test compares an output to a golden expected output, that is usually stored as `<generatedfilename>.expected`.  So `<mkFoo>.out.expected` would be the expected output of simulation for the module `mkFoo`.

In your case, if we look in `Testbench.bsv.bsc-vcomp-out`, we can see the failure:
```
checking package dependencies
compiling ./Dut.bsv
code generation for mkDut starts
Verilog file created: mkDut.v
Elaborated module file created: mkDut.ba
compiling ./Input.bsv
compiling ./Vectors.bsv
compiling Testbench.bsv
code generation for mkTestbench starts
Error: "Vectors.bsv", line 460, column 6: (S0015)
  Bluespec evaluation-time error: index 10 out-of-range (array update)
  During elaboration of the body of rule `set' at "Testbench.bsv", line 17,
  column 7.
  During elaboration of `mkTestbench' at "Testbench.bsv", line 9, column 8.
```
In this case, the test setup is a little complicated (and non-standard).  In this directory, the source BSV file is not handwritten, it is generated by a perl script in the subdirectory `generate`.  The test is randomly generating some arithmetic, using the Verilog simulator to compute the answer, and then testing that BSC's genera [... cut ...]  the second `$top`, the tests pass:
```
-       "$top" "$top" {-no-inline-reg} "sys$top.v.out.expected"
+       "$top" "" {-no-inline-reg} "sys$top.v.out.expected"

-       "$top" "$top" {-inline-reg} "sys$top.v.out.expected"
+       "$top" "" {-inline-reg} "sys$top.v.out.expected"
```
Sorry for this inconvenience.  But thank you for reporting it!  I can submit a PR to update these.  (Although if anyone has opinions on which is the better way to rewrite `gen_arith.v`, that would be appreciated.)

quark17, 2023-06-10:
Ah, the `Arith` failure is not because statements after the `$finish` are executing; it's because `$finish` isn't called with `0` as the argument, indicating that nothing should be printed, and IVerilog has changed its behavior when no argument is given, from printing nothing to now printing a message:
```
gen_arith.v:314: $finish called at 120 (1s)
```
And the perl script that creates a BSV file from the IVerilog output is not expecting that extra line.  Adding `(0)` to the `$finish` call makes it compile.  Or filtering out the extra line in the perl script.

Vekhir, 2023-06-10:
Wow, that's a quite detailed explanation of every step needed to find the issue, way more than I was expecting. And yet (or because of it), I am still able to follow it. Thank you very much, that surely took a lot of time!

I think your approach with just adding the `(0)` to finish is fine. I already tested the linked PR and for tests it fixes, it does do so. And it looks clean enough.

quark17, 2023-06-10:
Thanks.  There is also an unexpected pass with IVerilog 12:
```
Running /path/bsc/testsuite/bsc.evaluator/dynamic/strings/dynamic_strings.exp ...
XPASS: `sysStringInteger.v.out' identical to `sysStringInteger.out.expected' (PRMS iverilog)
```
And I ran the testsuite myself, with a development 13.0, and found an additional XPASS:
```
XPASS: `sysStringIntegerWithNull.v.out' identical to `sysStringIntegerWithNull.out.expected' (PRMS iverilog modelsim)
```

PR #562 (2023-06-11): Fix tests that fail with IVerilog 12+
This fixes #561 by addressing the two test directories that fail with IVerilog v12 (and 13.0 development at the HEAD of the iverilog repo).
closed 2023-06-11 by quark17