B-Lang-org/bsc #583: ICE "inlining not complete"

timksf, 2023-07-30:
Compiling the following code:
```Bluespec
package Test;

typedef 200 X;

module mkTest#(parameter UInt#(8) x)();

    UInt#(7) y = truncate(fromInteger(valueof(X)) / x);

    Reg#(UInt#(7)) r <- mkReg(y);

endmodule

endpackage
```
with `bsc -verilog -g mkTest -u Test.bsv` throws an ICE:
```
Bluespec Compiler, version 2023.01 (build 52adafa5)
Params.isConstAExpr: inlining not complete: _200_QUOT_100_QUOT_x__q1
```

quark17, 2023-07-31:
Thanks for reporting this! And sorry that you ran into it.  I note that you can avoid the failure by giving the hidden flag `-keep-add-size`.

The error is in a sanity check that BSC does after most of the Verilog backend stages, to check for example that the bit sizes of all the signals and expressions are still correct.  This failure is reported during the sanity check after the `veriquirks` stage (seen by running `bsc -v` to display the stages as BSC enters and exits them).  The `veriquirks` stage adjusts the representation to account for quirks in the Verilog language: for example, some Verilog operators can only be applied to a signal name and not to an expression; so this stage creates new signal names (and their assignments) to replace the expressions in such situations.

In your example, the value that is assigned to register `r` on reset is `(8'd200 / x) [6:0]`.  But Verilog doesn't allow range selection on expressions, it has to be on a signal name, so BSC has lifted the expression to a signal:
```
wire [7 : 0] _200_QUOT_x__q1;
...
assign _200_QUOT_x__q1 = 8'd200 / x ;
...
always@(posedge CLK)
  ...
  r <= _200_QUOT_x__q1[6:0];
  ...
```

In `bsc.hs`, in the code that steps through the stages, I see [this](https://github.com/B-Lang-org/bsc/blob/98a5797a571818445ed7a4e7b898825d089099d3/src/comp/bsc.hs#L1156-L1164) for `veriquirks`:
```
-- Transform to adapt to Verilog quirks
start flags DFveriquirks
let aqmod = aVeriQuirks flags asynmod
-- this check is too strict on plus operator output size
-- XXX fix the check, don't disable it
when (not (keepAddSize flags))
     (asCheck flags aqmod "veriquirks")
t <- dump errh flags t DFveriquirks dumpnames aqmod
stats flags DFveriquirks aqmod
```
There is a guard on the call to `asCheck` (the function that performs the sanity check), to not call the check if the flag `-keep-add-size` is used.  Thus, you can avoid the error by using that flag.  However, as the `XXX` comment says, we should fix the check -- both for your example and for when `-keep-add-size` is used.

The `-keep-add-size` flag is used in the `veriquirks` stage.  In BSV/BH, the addition and subtraction operators for bit vectors is defined to operate on two vectors of the same size; so the smaller vector is often zero-extended.  This flag tells BSC to remove the zero-extend (and rely on Verilog's size-inference rules).  (This is a flag from BSC's very first days, and I don't think anyone has used it since then?)

Anyway, the failure is in `ACheck.hs`, in `chkAVInst` which is the function for checking the submodule instantiations.  It performs a check that the arguments to submodule parameters are static (constant) expressions (by calling `isConstAExpr` which is defined in `Params.hs`).  The function `isConstAExpr` expects the expression to be fully inlined and to not refer to wire signals -- but in this example, the expression was lifted to a signal, so that check fails.

Register submodules are inlined by default, so it's hard to see the issue in the output Verilog, but we can provide the flag `-no-inline-reg` to keep the register as a submodule instantation.  However, if we do that, BSC aborts in a later stage, again complaining about the bad submodule parameter.  If it had generated the Verilog, it would look like this:
```
module mkTest (...);
  parameter [7 : 0] x = 8'b0;
  ...
  assign _200_QUOT_x__q1 = 8'd200 / x ;
  ...
  RegN #(.width(32'd7), .init(_200_QUOT_x__q1[6:0])) r(...);
  ...
endmodule
```
Offhand, I don't know if this is legal Verilog code?  Can a continuously-assigned signal be used in the argument to a parameter?  Perhaps the solution here is for BSC to remove the range-selection (and allow Verilog's size-inference to perform the truncate) and therefore no lifting of the expression is needed.  However, a more complicated use of parameters in BSV/BH could be devised that wouldn't be so easily fixed (for example, if the extraction occurs inside a nested expression and not at the outer level).

If there are situations that can't be represented in Verilog, BSC would need to give a user error.  But it should make exceptions for when it becomes possible due to inlining of the module.

mieszko, 2023-08-25:
documenting offline discussion:

```

mieszko, 2023-08-25:
documenting offline discussion:

using `localparam` compiles; for example this:

```
module mkTest ();
  parameter [7 : 0] x = 8'b0;
  localparam foo = 8'd200 / x;
  localparam bar = foo[6:0];
  wire          CLK, RST, EN;
  wire [6:0] Q_OUT, D_IN;
  RegN #(.width(32'd7), .init(bar)) r(.EN(EN), .RST(RST), .Q_OUT(Q_OUT), .D_IN(D_IN));
endmodule
```

the trick here is to factor out `foo` and use it to define `bar` because otherwise the Verilog parser chokes on the `[`.