B-Lang-org/bsc #586: Potential Bug in StmtFSM for-loop parsing

yannickl96, 2023-08-03:
Hello there!

Today I encountered a compiler error (version 2023.01) when using a vector of registers for loop iteration counting in StmtFSMs. Please see below reproducing example:
```
package MWRE;
    import StmtFSM::*;
    import Vector::*;

    module mkMWRE(Empty);
        Vector#(2, Reg#(int)) vec <- replicateM(mkRegU);
        Reg#(int) foo = vec[0];
        
        Stmt s = seq
            par
                for(foo <= 0; vec[0] < 10; foo <= vec[0] + 1) action
                    $display("vec[0] is %d", vec[0]);
                endaction

                for(vec[1] <= 0; vec[1] < 10; vec[1] <= vec[1] + 1) action
                    $display("vec[1] is %d", vec[1]);
                endaction
            endpar
        endseq;

        mkAutoFSM(s);
    endmodule
endpackage
```
and the corresponding compiler message:
```
$ bsc -u -sim -g mkMWRE MWRE.bsv 
checking package dependencies
Error: "MWRE.bsv", line 14, column 24: (P0128)
  Array element assignments forbidden in expression block context
```
According to the Library Reference Guide StmtFSM grammar, we have the following relevant productions (shortened):
```
fsmStmt          ::= exprFsmStmt | forFsmStmt

exprFsmStmt ::= regWrite ; | expression ;

forFsmStmt    ::= for ( fsmStmt ; expression ; fsmStmt )
                               loopBodyFsmStmt
```
and the corresponding productions for a *regWrite* (taken from the Language Reference Guide):
```
regWrite ::= lValue <= expression
lValue    ::= identifier
              |    lValue . identifier
              |    lValue [ expression ]
              |    lValue [ expression : expression ]
```
Please correct me, if I am wrong. But to my understanding the parser should in theory be able to identify the two assignments in line 14 of my MWRE as regWrites.

Thank you for your time!

quark17, 2023-08-04:
I don't see a reason why that shouldn't be accepted.  Thank you for reporting the issue!

As you saw, one workaround is to replace the vector selection with a variable name.  Another is to wrap the selection with `asReg` or `asIfc`.

I suspect that fixing this might be as simple as adding the following clause to the flags used for parsing the init and increment parts of the for-loop control triple:
```
allowSubscriptAssign = (ctxt == ISCSequence)
```
This is in the function `pImperativeForControl` ([here](https://github.com/B-Lang-org/bsc/blob/fdf107c7ed9419e23db251dc9ec1b5527e4a543b/src/comp/Parser/BSV/CVParser.lhs#L2520)) which parses the triple.  It's used in `pImperativeFor`, defined above it, that parses the for loop.

The question is whether this would allow something incorrect to parse -- probably not?  Note that it would also allow assignment to a range, not just a single index:
```
Reg#(Bit#(32)) baz <- mkReg('1);
...
    for(baz[7:0] <= 0; baz[7:0] < 10; baz[7:0] <= baz[7:0] + 1) action
      $display("baz[7:0] is %h", baz[7:0]);
      $display("baz is %h", baz);
    endaction
```

quark17, 2023-08-04:
Having a look at the other fields in the type `ImperativeFlags`,  I see that we might also want to assert `allowFieldAssign`, to allow code like this:
```
for(rg_s.f1 <= 0; rg_s.f1 < 10; rg_s.f1 <= rg_s.f1 + 1) action
  $display("rg_s.f1 is %h", rg_s.f1);
endaction
```

quark17, 2023-08-04:
I've merged a PR that fixes this and adds test cases.  Thanks again!

closed 2023-08-04 by quark17