B-Lang-org/bsc #599: YICES_STUB option leads to build failure

Vekhir, 2023-08-13:
### Issue
When using the build option YICES_STUB=1 during building in order to disable Yices, BSC fails to build with error

```
/build/bluespec-git/src/bsc/inst/bin/bsc -stdlib-names -bdir /build/bluespec-git/src/bsc/build/bsvlib -p . -vsearch /build/bluespec-git/src/bsc/build/bsvlib -no-use-prelude Prelude.bs
Error: Command line: (S0081)
  The flag `-sat-yices' was used, but a proper shared object file was not
  found. Please specify a different SAT solver or check that the
  LD_LIBRARY_PATH or BLUESPEC_LD_LIBRARY_PATH includes a valid `libyices.so.2'
  file.
```

Apparently, building with Yices disabled still requires Yices to build? If I had to guess, then probably the logic to choose the default solver fails to switch to STP if Yices isn't available.

### Additional information
Error occurs on the main branch (commit 0b385553a71d45129088274b61a4f5be59090d2c). Omitting the build option builds just fine.

Command: `make YICES_STUB=1 GHC="ghc -dynamic" GHCJOBS=4 GHCRTSFLAGS='+RTS -M5G -A128m -RTS' install-src`

### System information
OS: Arch Linux
Kernel: 6.4.9-arch1-1
GHC: 9.0.2-3
BSC: 0b385553a71d45129088274b61a4f5be59090d2c (via [bluespec-git](https://aur.archlinux.org/packages/bluespec-git))

<details>
<summary><b>Full error log</b></summary>

```
make  -C src  PREFIX=/build/bluespec-git/src/bsc/inst  install
make[1]: Entering directory '/build/bluespec-git/src/bsc/src'
make  -C vendor/stp   PREFIX=/build/bluespec-git/src/bsc/inst  install
make[2]: Entering directory '/build/bluespec-git/src/bsc/src/vendor/stp'
make -C src install
make[3]: Entering directory '/build/bluespec-git/src/bsc/src/vendor/stp/src'
make -C AST 
make -C sat core
make[4]: Entering directory '/build/bluespec-git/src/bsc/src/vendor/stp/src/AST'
make[4]: Entering directory '/build/bluespec-git/src/bsc/src/vendor/stp/src/sat'
make LIB="core" -C core libr
Making dependencies
make[5]: Entering directory '/build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core'
Making dependencies
Compiling: /build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core/Solver.or
Compiling: utils/Options.or
Compiling: utils/System.or
../mtl/template.mk:71: warning: pattern recipe did not update peer target '../utils/System.od'.
../mtl/template.mk:71: warning: pattern recipe did not update peer target '../utils/System.op'.
../mtl/template.mk:71: warning: pattern recipe did not update peer target '../utils/System.o'.
../mtl/template.mk:71: warning: pattern recipe did not update peer target '../utils/Options.od'.
../mtl/template.mk:71: warning: pattern recipe did not update peer target '../utils/Options.op'.
../mtl/template.mk:71: warning: pattern recipe did not update peer target '../utils/Options.o'.
./genkinds.pl
Making dependencies
../mtl/template.mk:71: warning: pattern recipe did not update peer target '/build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core/Solver.od'.
../mtl/template.mk:71: warning: pattern recipe did not update peer target '/build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core/Solver.op'.
../mtl/template.mk:71: warning: pattern recipe did not update peer target '/build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core/Solver.o'.
Making library: libcore_release.a ( /build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core/Solver.or utils/Options.or utils/System.or )
a - /build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core/Solver.or
a - ../utils/Options.or
a - ../utils/System.or
Making Soft Link: libcore_release.a -> libcore.a
make[5]: Leaving directory '/build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core'
make LIB="core_prop" -C core_prop libr
make[5]: Entering directory '/build/bluespec-git/src/bsc/src/vendor/stp/src/sat/core_prop'
Making dependencies

[Building STP...]

make[4]: Leaving directory '/build/bluespec-git/src/bsc/src/vendor/stp/src/sat'
[... 341 lines cut ...]
mkdir -p -m 755 /build/bluespec-git/src/bsc/inst/bin
install -m 755 wrapper.sh /build/bluespec-git/src/bsc/inst/bin/bluetcl
make[2]: Leaving directory '/build/bluespec-git/src/bsc/src/comp'
make  -C Libraries  PREFIX=/build/bluespec-git/src/bsc/inst  install
make[2]: Entering directory '/build/bluespec-git/src/bsc/src/Libraries'
make -C Base1 build &&  make -C Base2 build &&  make -C Base3-Misc build &&  make -C Base3-Contexts build &&  make -C Base3-Math build && true
make[3]: Entering directory '/build/bluespec-git/src/bsc/src/Libraries/Base1'
mkdir -p /build/bluespec-git/src/bsc/build/bsvlib
/build/bluespec-git/src/bsc/inst/bin/bsc -stdlib-names -bdir /build/bluespec-git/src/bsc/build/bsvlib -p . -vsearch /build/bluespec-git/src/bsc/build/bsvlib -no-use-prelude Prelude.bs
Error: Command line: (S0081)
  The flag `-sat-yices' was used, but a proper shared object file was not
  found. Please specify a different SAT solver or check that the
  LD_LIBRARY_PATH or BLUESPEC_LD_LIBRARY_PATH includes a valid `libyices.so.2'
  file.
make[3]: *** [Makefile:26: /build/bluespec-git/src/bsc/build/bsvlib/Prelude.bo] Error 1
make[3]: Leaving directory '/build/bluespec-git/src/bsc/src/Libraries/Base1'
make[2]: *** [Makefile:31: build] Error 2
make[2]: Leaving directory '/build/bluespec-git/src/bsc/src/Libraries'
make[1]: *** [Makefile:62: install] Error 2
make[1]: Leaving directory '/build/bluespec-git/src/bsc/src'
make: *** [GNUmakefile:41: install-src] Error 2

```

</details>

Vekhir, 2023-08-13:
And for similar reasons as #595, the option won't work on macOS at all, because no `.dylib` is created, only `.so`.

quark17, 2023-08-14:
There is no logic to choose the default (it's always Yices) and there is no fall-back if it's not available -- see the message for commit 49760f3a.  That means that a build with Yices stub would need to have `-sat-stp` explicitly provided every time.  That does seem unfriendly.

The check for the solver (and error if it's a stub) is done in `checkSATFlags` in `SAT.hs`.  We could change that function to try other solvers if a stub is found.  We could change the type of `satBackend` in `Flags` to be a `Maybe` that starts as `Nothing`, so that it captures whether a solver was specifically requested.  In theory, the build of `bsc` could consult the makefile flags and hardcode STP as the default, if Yices is built as a stub; but I'm not sure I want to go that route.

Vekhir, 2023-08-14:
I'm not sure we are on the same page.

The issue isn't so much that I'd have to pass `-sat-stp` everywhere to use BSC, rather, I cannot use it at all because the flag isn't passed during the build process of this very library itself. 

So using YICES_STUB means that there is no package at all, because the build fails.