B-Lang-org/bsc #614: for make install-src i got this error can anyone help for this

Rosith111, 2023-09-16:
Makefile:294: recipe for target 'bsc' failed
make[2]: *** [bsc] Error 1
make[2]: Leaving directory '/home/roshit/Tools/bsc/src/comp'
Makefile:57: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/roshit/Tools/bsc/src'
GNUmakefile:41: recipe for target 'install-src' failed
make: *** [install-src] Error 2

quark17, 2023-09-16:
Is there text before this that you haven't shown?  This is what the `make` command reports after the error, and doesn't show the error itself.

Rosith111, 2023-09-16:
<b> Thank you for the Response  </b>  <br>
after cloning the bsc repository from git. i got some error.
can you guide how to install bsc in my system

quark17, 2023-09-16:
I may be able help, but you need to provide more information.  What error message are you getting?

Rosith111, 2023-09-19:
this is what shown when make install-src 

roshit@ubuntu:~/Tools/bsc$ sudo make install-src
make  -C src  PREFIX=/home/roshit/Tools/bsc/inst  install
make[1]: Entering directory '/home/roshit/Tools/bsc/src'
make  -C vendor/stp   PREFIX=/home/roshit/Tools/bsc/inst  install
make[2]: Entering directory '/home/roshit/Tools/bsc/src/vendor/stp'
make -C src install
make[3]: Entering directory '/home/roshit/Tools/bsc/src/vendor/stp/src'
mkdir -p ../lib
mkdir -p ../include
cp -df lib/libstp.so.1 lib/libstp.so ../lib
cp -df c_interface/*.h ../include
mv ../include/c_interface.h ../include/stp_c_interface.h
make[3]: Leaving directory '/home/roshit/Tools/bsc/src/vendor/stp/src'
ln -fsn HaskellIfc include_hs
install -m 755 -d /home/roshit/Tools/bsc/inst/lib/SAT
install -m 644 lib/libstp.so.1 /home/roshit/Tools/bsc/inst/lib/SAT
make[2]: Leaving directory '/home/roshit/Tools/bsc/src/vendor/stp'
make  -C vendor/yices PREFIX=/home/roshit/Tools/bsc/inst  install
make[2]: Entering directory '/home/roshit/Tools/bsc/src/vendor/yices'
make -C v2.6 install
make[3]: Entering directory '/home/roshit/Tools/bsc/src/vendor/yices/v2.6'
(cd yices2 ; \
	autoconf ; \
	 ./configure --prefix=/home/roshit/Tools/bsc/src/vendor/yices/v2.6/yices2-inst ; \
	make ; \
	make install \
	)
checking build system type... x86_64-pc-linux-gnu
checking whether make sets $(MAKE)... yes
checking for a sed that does not truncate output... /bin/sed
checking whether ln -s works... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for a BSD-compatible install... /usr/bin/install -c
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for gcc option to accept ISO C99... none needed
checking for ranlib... ranlib
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ld... ld
checking for ar... ar
checking for strip... strip
checking for gperf... gperf
checking gperf version... 3.1
checking whether gcc accepts option -fno-stack-protector... yes
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking whether byte ordering is bigendian... no
configure: Trying to find a usable libgmp.a
configure: Searching in  /usr/local/lib/x86_64-linux-gnu /lib/x86_64-linux-gnu /usr/lib/x86_64-linux-gnu /usr/lib/x86_64-linux-gnu64 /usr/local/lib64 /lib64 /usr/lib64 /usr/local/lib /lib /usr/lib /usr/x86_64-linux-gnu/lib64 /usr/x86_64-linux-gnu/lib /usr/local/lib /usr/lib /lib
checking for libgmp.a in /usr/local/lib/x86_64-linux-gnu... no
checking for libgmp.a in /lib/x86_64-linux-gnu... no
[... 375 lines cut ...]
# we need to build targets from here sequentially, as they operate in the same workspace
make  -C comp -j1   PREFIX=/home/roshit/Tools/bsc/inst  install
make[2]: Entering directory '/home/roshit/Tools/bsc/src/comp'
Using tclsh: /usr/bin/tclsh
Using tcl include flags: -I/usr/include/tcl8.6
Using tcl library flags: -ltcl8.6
Building with GHC 8.0.2
----- Normal build options -----
bsc start Tue Sep 19 07:43:00 PDT 2023
./update-build-version.sh
BuildVersion.hs up-to-date
./update-build-system.sh
BuildSystem.hs up-to-date
ghc -Wtabs -hidir /home/roshit/Tools/bsc/src/comp/../../build/comp -odir /home/roshit/Tools/bsc/src/comp/../../build/comp -stubdir /home/roshit/Tools/bsc/src/comp/../../build/comp -main-is Main_bsc \
	-O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -package text  -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp'  -I../vendor/stp/include -I../vendor/yices/include -L../vendor/htcl -I/usr/include/tcl8.6  -lpthread --make bsc -j1 +RTS -M4G -A128m -RTS "-with-rtsopts=-H256m -K10m -i1" -rtsopts -L../vendor/stp/lib -lstp -L../vendor/yices/lib -lyices 
<command line>: cannot satisfy -package text
    (use -v for more information)
Makefile:294: recipe for target 'bsc' failed
make[2]: *** [bsc] Error 1
make[2]: Leaving directory '/home/roshit/Tools/bsc/src/comp'
Makefile:57: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/roshit/Tools/bsc/src'
GNUmakefile:41: recipe for target 'install-src' failed
make: *** [install-src] Error 2

quark17, 2023-09-19:
The error is this line:
```
cannot satisfy -package text
```
You are using GHC version 8.0, as shown on this line:
```
Building with GHC 8.0.2
```
I would expect that version to have the `text` package, so I am surprised by the error.  You can try installing the package with `cabal`, in the same was as the other packages described in the build instructions:
```
cabal v1-install text

# Or this for older cabal v2
cabal install text
```
To see what packages are currently installed, you can use this command:
```
ghc-pkg list
```
Alternatively, you can try using a newer version of GHC, such as 9.0.2 or 9.4.7, which can be installed using `ghcup` as mentioned in the build instructions.

Rosith111, 2023-09-19:
can i know the command for GHC 9.0.2 or 9.4.7 i am unable to get from google

quark17, 2023-09-19:
Use `ghcup` to install specific versions of GHC.  The website is: https://www.haskell.org/ghcup/

This information is given in the instructions for building BSC, in the file `INSTALL.md`: https://github.com/B-Lang-org/bsc/blob/main/INSTALL.md

closed 2023-09-23 by quark17