B-Lang-org/bsc #623: i got several error onthe installation process

Rosith111, 2023-10-07:


quark17, 2023-10-07:
This is the same error that you reported in issue #614 and I gave an answer there.  What OS are you using? How did you install GHC?  And how did you install the other libraries (regex-compat, syb, old-time, and split)?

To repeat what I said in #614 , the error is this line:
> cannot satisfy -package text

This is a message from GHC, saying that the package `text` is not installed.  The version of GHC is shown in this line:
> Building with GHC 8.0.2

That version of GHC should have the `text` library by default, but if not, you can try installing it, in the same way that you installed the other packages.  For example, if you used this command to install the packages:
```
apt-get install \
    libghc-regex-compat-dev \
    libghc-syb-dev \
    libghc-old-time-dev \
    libghc-split-dev
```

Then do this command, to install the `text` package:
```
apt-get install ibghc-text-dev
```

If you used one of these commands to install the packages:
```
cabal install regex-compat syb old-time split

cabal v1-install regex-compat syb old-time split
```

Then do the equivalent for the `text` package:
```
cabal install text

cabal v1-install text
```

Alternatively, you can try upgrading to a newer version of GHC (which you can install with GHCUP).  However, it would help to know what OS you a building on, and what mechanism you used to install GHC and to install the additional libraries.

Rosith111, 2023-10-10:
sorry for the inconvenience as u already said about it but due to some technical error i was not getting ghc 9.0.2 or 9.4.7 due to that again i got same error on the install-src

yeah I was using UBUNTU 18.04.6 version 

and to install GHC I was followed the script which was given in the bsc repository  i.e.,
**install.md**

quark17, 2023-10-10:
If you have installed a new GHC with GHCUP, then you need to rebuild the libraries for that version, using this command:
```
cabal v1-install regex-compat syb old-time split
```
Does that work?

The error message indicates that libraries are missing.  You probably skipped that step before.  There is not one script in `INSTALL.MD`, there are multiple options.  For installing the libraries, you can either execute something like this:
```
apt-get install \
    libghc-regex-compat-dev \
    libghc-syb-dev \
    libghc-old-time-dev \
    libghc-split-dev
```
or do something like this:
```
cabal v1-install regex-compat syb old-time split
```
If you have used GHCUP to install GHC, then you should use the second option (`cabal`).

I would guess that you skipped this step the first time.  And, in any case, after installing a new GHC with GHCUP, you need to re-run the `cabal` command.

Rosith111, 2023-10-10:
thank you for taking a part of our installation by installing the text package it is sucessfully installed...

thank you very much

closed 2023-10-10 by Rosith111