B-Lang-org/bsc #641: Method with internal input-dependent "if" mistakenly scheduled as SBR to self

dramforever, 2023-11-06:
Given this module

```bsv
interface Meow;
    method Action meow(Int#(32) value);
endinterface

(* synthesize *)
module mkPositiveMeow(Meow);
    Reg#(Int#(32)) r <- mkReg(0);

    rule disp;
        $display(r);
    endrule

    method Action meow(Int#(32) value);
        if (value > 0) begin
            $display("actually set");
            r <= value;
        end
    endmethod
endmodule
```

Semantically, `meow(1)` then `meow(-1)` results in `r` being set to `1`, not `-1`, which means `meow C meow`. However BSC infers `meow SBR meow`:

```
=== Generated schedule for mkPositiveMeow ===

Method schedule
---------------
Method: meow
Ready signal: True
Sequenced before (restricted): meow
 
Rule schedule
-------------
Rule: disp
Predicate: True
Blocking rules: (none)
 
Logical execution order: disp, meow

==============================================
```

The incorrect behavior can be seen with this testing module:

```bsv
(* synthesize *)
module mkMeowTwice();
    Meow meow <- mkPositiveMeow;

    (* fire_when_enabled *)
    (* execution_order = "meow1, meow2" *)
    rule meow1;
        $display("set 1");
        meow.meow(1);
    endrule

    (* fire_when_enabled *)
    rule meow2;
        $display("set -1");
        meow.meow(-1);
    endrule
endmodule
```

If using `-sim`, or `-verilog` but with `(* synthesize *)` on `mkPositiveMeow` removed, `r` is correctly set to `1` as expected:

```
          0
set 1
actually set
set -1
          1
set 1
actually set
set -1
...
```

With `(* synthesize *)` on `mkPositiveMeow` and `-verilog`, `meow1` and `meow2` are somehow still both schedulable in the same cycle, but `r` incorrectly stays at `0`

```
set 1
set -1
          0
          0
set 1
set -1
set 1
set -1
          0
          0
...
```

Inspecting the generated `mkMeowTwice.v` reveals that it has indeed incorrectly merged `meow(1)` and `meow(-1)` into `meow(-1)` based on the incorrectly inferred `meow SBR meow`:

```verilog
  // submodule meow
  mkPositiveMeow meow(.CLK(CLK),
		      .RST_N(RST_N),
		      .meow_value(meow$meow_value),
		      .EN_meow(meow$EN_meow),
		      .RDY_meow());

  // submodule meow
  assign meow$meow_value = 32'hFFFFFFFF ;
  assign meow$EN_meow = 1'b1 ;
```

bsc version: 2023.07

quark17, 2023-11-10:
Thanks for reporting this!  I have not had a chance to look at it yet, but will get to it.  If the determination of SBR isn't taking into account the conditions on method calls, that seems like a bug.  It's surprising to still find a scheduling bug in BSC!  And a priority issue to fix.

I wonder if there are any corner cases to consider.  For example, if the argument is written to a submodule that has a combinational output (like a wire module) and that output is used as condition -- but maybe in that case, the write/read of the combinational value would cause a conflict anyway.

sequencer, 2023-12-02:
any update on this issue?

quark17, 2023-12-25:
In `ASchedule`, there is a section of code with this comment:
> ActionValue methods with arguments should be considered to conflict with themselves if the return value uses an argument

This code lazily constructs a map (`port_usemap`) from a def ID to the set of module ports that appear in its expression.  Then, for every ActionValue method, it finds the def for the return value and looks it up in the map, and checks whether the method's argument is one of the ports in the set.  If so, it introduces an edge into the conflict graphs that are being built.

This is similar to the situation that we want to check for: if an argument to an Action or ActionValue method appears in the condition of any of the actions.  We can share the `port_usemap` structure (to avoid duplicating that computation).  But I can imagine different ways of using this info, and I'm not yet sure which has the best efficiency tradeoffs.

An initial conflict graph is constructed by considering all pairs of rules/methods, including the rule or method with itself, and for each pair that could execute together (their conditions are not mutually exclusive), look at each pair of method calls (one from rule A and one from rule B) and test if they conflict.  To this initial graph, other conflicts are then added.  For example, the code mentioned above -- checking for arguments in the ActionValue return value -- computes new edges to add to the graph.

For a check of arguments in the action conditions, we could compute edges to add in after.  This would require looping through every action of every action method of the module, and looking it up in the `port_usemap`.  If there are lots of actions and lots of conditions, this could grow (and force a lot of compuation of the map).  However, we could first do a lookup of the method in the initial graph, to see if the method was already determined to conflict with itself; if so, then we could skip it, and only do the work of checking any method that was SBR (in the initial graph).  That is adding extra work (a lookup), which might not save much in situations where the number of actions with conditions is small.  Although, I guess one lookup per method is not a huge cost -- so maybe we should do that.

Alternatively, we could consult the condition while constructing the initial graph.  If we did the argument-in-action-condition check first, and it determined that the method conflicts with itself, then we could avoid checking all the pairwise uses.  But I think that's the less common case; better to check the pairwise uses first and then look at the conditions if necessary.  That's the same order as computing the initial graph and then adding edges later, but if we do it during the construction of the initial graph then we save the lookup (assuming the compiler isn't smart enough to fuse those two computations).  I think it's better to keep it as a separate computation afterwards (and not further convolute the construction of the initial graph).  We can always start with that implementation and then adjust it later if profiling shows it to be a performance issue.  So I'll look at doing that.

quark17, 2023-12-26:
I have a candidate fix in the [`issue-641-sbr-cond` branch](https://github.com/quark17/bsc/tree/issue-641-sbr-cond) of my repo.  I'm unsure of the performance implications though -- not the ones discussed above, but the fact that BSC was forcing evaluation of the conflict graphs, and my commits rearrange the order of computation and I'm unsure where the forcing should go (if it's even still needed?) in the new order.

The original code computed the new edges to add to the scheduling graph for user pragmas and for action value methods with arguments, then it constructed the initial CF graph (based only on the method calls) and then added the extra edges to it and then forced evaluation of that result; then it did the same with the PC graph, and then with the SC graph.  My new code rearanges things: The initial CF graph is computed, the initial PC graph is computed, the initial SC graph is computed, and then the extra pragma edges are computed and added, and then the extra edges for action methods are computed and added.  I'm unsure where to put the forcing in this new order (if at all).

The reorganization is needed because I want to short-circuit the checking of action methods by first looking up in the graph whether `C` was already inferred for the method (and if so then skip it and move on).  If we don't do that lookup, then the original order was fine, and the forcing can stay where it was.

I settled for forcing evaluation in two places: as each initial CF, PC, and SC graph is computed, it is forced; and then after all the extra edges are added, the result is forced again.  I don't have a sense of whether that's adding extra time to the compile (for designs with large graphs / lots of rules and methods).  FYI, the forcing was added in Feb 2010, so GHC would have still been version 6 at the time (and possibly not even 6.12).

I'll have to think on it, or do some more experiments.  If I'm lucky, maybe I can find the example from 2010 that was profiled to identify the laziness leak.  It was a crossbar with 81 interfaces (of 14 methods each), and BSC was not scaling well with large numbers of methods, so a number of fixes were made.  The design also went through various iterations (including 81 x 81 rules) and I don't know if I can find the specific version that led to the forcing of the conflict graph.  But I think the idea was that the conflict graph was a thunk that took up a lot of memory (and eventually max out the system memory), and I think that it probably is good to force each graph, so that there aren't three in memory at the same time, but I didn't know if forcing just the initial graphs is enough or if there's a reason to also force evaluation after the extra edges are added.

PR #656 (2023-12-30): Fix scheduling annotations for methods with actions conditional on the arguments
Fixes #641.

There is some reorganization of the code in `ASchedule` that constructs the three conflict graphs (CF, SC, and PC, indicating the scheduling limitations between rules/methods).  This code contained forced evaluations of the graphs, to prevent memory usage from growing too much (due to the graphs being contructed lazily), and the location of that forcing has moved -- the graphs are constructed in steps (first the initial graph based on pairwise method calls then user pragmas then methods with arguments [... cut]
closed 2023-12-30 by quark17