B-Lang-org/bsc #663: `-dparsed` produces invalid BSV module function that takes another module as an argument

RyanGlScott, 2024-01-14:
Given this BSV code:

```bsv
package Foo;

module [Module] helloWorld#(Module#(Empty) m)(Empty);
  let e <- m;
endmodule

endpackage
```

Compiling it with `bsc -dparsed=FooParsed.bsv Foo.bsv` yields:

```bsv
package Foo;
module helloWorld#(Module#(Empty) m)(Empty);
  let e <- m;
endmodule: helloWorld

endpackage: Foo
```

This is almost the same code, but there is one key difference: the `module` keyword lacks the `[Module]` part after it. This change is significant enough to cause bsc to reject it:

```
$ bsc FooParsed.bsv 
Warning: Unknown position: (S0001)
  File name `FooParsed.bsv' does not match package name `Foo'
Error: "FooParsed.bsv", line 2, column 8: (T0029)
  Signature mismatch (given too general):
  given:
  function a__#(Empty) f(Module#(Empty) x1)   provisos (IsModule#(a__, b__))
  deduced:
  module f#(Module#(Empty) x1)(Empty)   provisos (IsModule#(Module, Id__))
```

Any reason not to preserve the `[Module]` part?

quark17, 2024-01-17:
Thanks for reporting these.  There's been no effort to make sure that the pretty-printer for an entire BSV file will read back in correctly and identically, so I am not surprised if you find issues like this.  The only concern was the printing of any snippets that appear in error messages to the user, which are usually just types, or IDs, or individual method calls.  The `-d` dump flags were considered hidden features for developers, and they didn't need to be perfect (and could change or disappear at any time).  But it's a good aim to have the pretty-printer work for full programs, so thank you for reporting this.

The module type should be preserved.  I assume that this is just a bug in `CVPrint.hs`, probably around line 165, in the function `p2defs`:
```
line1 = t"module" <+> pvpId d i <> f ys <> t"(" <> pvPrint d 0 ity <> t")"
```

RyanGlScott, 2024-01-18:
Thanks! And I certainly don't expect the pretty-printer to be 100% accurate in all cases. My use case is that I am generating very simple BSV code programmatically using the `bsc` source code, and I ran into some situations where even very simple code fails to parse when fed back into `bsc`. My goal is to fix only those issues that I encounter when pretty-printing the very limited subset of BSV that I'm generating.

I'll give `CVPrint` a closer look. For posterity's sake, here is a permalink to the code mentioned in https://github.com/B-Lang-org/bsc/issues/663#issuecomment-1897040650: 

https://github.com/B-Lang-org/bsc/blob/e4361d9177f19733ce8eeafd27737e10bd7c58a1/src/comp/CVPrint.hs#L165