B-Lang-org/bsc #669: Testsuite reports an error getting BSC version

krame505, 2024-01-31:
I'm having some trouble getting the bsc testsuite to run properly.  Running `make check-suite` at the top level ends in a failure
```
...
cleaning /home/lucas/hardware/bsc/testsuite/bsc.verilog/tasks/interfacecalls
cleaning /home/lucas/hardware/bsc/testsuite/bsc.verilog/tasks/plusargs
cleaning /home/lucas/hardware/bsc/testsuite/bsc.verilog/tasks/real
cleaning /home/lucas/hardware/bsc/testsuite/bsc.verilog/tasks/time
cleaning /home/lucas/hardware/bsc/testsuite/bsc.verilog/undet
cleaning /home/lucas/hardware/bsc/testsuite/bsc.verilog/v95
cleaning /home/lucas/hardware/bsc/testsuite/bsc.verilog/vcd
echo "Local check is not supported at top level" 
Local check is not supported at top level
MAKEFLAGS= BSCTEST=1 BSC=/home/lucas/hardware/bsc/testsuite/../inst/bin/bsc BSC_OPTIONS="" BSDIR=/home/lucas/hardware/bsc/testsuite/../inst/lib DUMPBO=/home/lucas/hardware/bsc/testsuite/../inst/bin/dumpbo BSC2BSV=/home/lucas/hardware/bsc/testsuite/../inst/bin/bsc2bsv VCDCHECK=/home/lucas/hardware/bsc/testsuite/../inst/bin/vcdcheck SHOWRULES=/home/lucas/hardware/bsc/testsuite/../inst/bin/showrules BLUESPECDIR=/home/lucas/hardware/bsc/inst/lib BSC_VERILOG_SIM=iverilog TEST_CONFIG_DIR=/home/lucas/hardware/bsc/testsuite/config BLUETCL=/home/lucas/hardware/bsc/testsuite/../inst/bin/bluetcl OSTYPE=Linux LC_ALL=en_US.UTF-8 SYSTEMC_INC= SYSTEMC_LIB= SYSTEMC_CXXFLAGS= PATH="/home/lucas/hardware/bsc/inst/lib/../bin:/home/lucas/opt/f4pga/xc7/conda/condabin:/home/lucas/opt/f4pga/xc7/install/bin/:/home/lucas/hardware/bsc/inst/bin/:/home/lucas/.cabal/bin:/home/lucas/.ghcup/bin:/home/lucas/bin:/usr/local/bin:/home/lucas/.cargo/bin:/home/lucas/.local/bin:/home/lucas/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin"  runtest --tool bsc  --objdir . --status 
Using ./lib/bsc.exp as tool init file.
Test run by lucas on Tue Jan 30 18:44:21 2024
Native configuration is x86_64-pc-linux-gnu

		=== bsc tests ===

Schedule of variations:
    unix

Running target unix
Using /usr/share/dejagnu/baseboards/unix.exp as board description file for target.
Using /usr/share/dejagnu/config/unix.exp as generic interface file for target.
Using ./config/unix.exp as tool-and-target-specific interface file.
testconfig dir is: /home/lucas/hardware/bsc/testsuite/config
Sourcing: /home/lucas/hardware/bsc/testsuite/config/verilog.tcl
testconfig dir is: /home/lucas/hardware/bsc/testsuite/config
Sourcing: /home/lucas/hardware/bsc/testsuite/config/bluetcl.tcl
ERROR: failed to execute bluetcl to get BSC version: Error: Unexpected additional arguments: "bsc".
make[1]: *** [/home/lucas/hardware/bsc/testsuite/suitemake.mk:99: check] Error 1
make[1]: Leaving directory '/home/lucas/hardware/bsc/testsuite'
make: *** [GNUmakefile:67: check-suite] Error 2
```

I get the same issue when trying to run the tests directly from the testsuite folder.

I'm guessing this is something to do with my system, since the CI is currently passing.  I'm on Ubuntu 22.04.

quark17, 2024-01-31:
You checked out a version of the repo/testsuite that is newer than the bsc/bluetcl that you have built.  Either roll back the testsuite or update your BSC.  In commit b59eb115, subcommands were added to the Bluetcl `version` command and used in the testsuite infrastructure.

krame505, 2024-01-31:
That was the issue - I apparently was using an old build of bsc.  Thanks.

closed 2024-01-31 by krame505