B-Lang-org/bsc #673: Internal Compiler Error when quoting a character value.

yseulp, 2024-02-15:
Error message: Internal Bluespec Compiler Error:
Please report this failure to the BSC developers, by opening a ticket
in the issue database: https://github.com/B-Lang-org/bsc/issues
The following internal error message should be included in your
correspondence along with any other relevant details:
Bluespec Compiler, version 2023.07-7-g527eaa0b (build 527eaa0b)
quoting a character value

quark17, 2024-02-15:
Thank you for reporting this. I apologize that you encountered the error.

To determine the cause, we need more information. Are you able to provide the source code that you compiled, such as Testbench.bsv and any files that it imports?

Also, is there more to the error message? After "quoting a character value", I would expect to see a number.

yseulp, 2024-02-15:
Hi, here is the source code I tried to compile. And there is no more error message. 

package MultTb;
import Mult::*; 
module mkTestbench (Empty); 

    Mult_ifc m1 <- mkMult;
    Mult_ifc m2 <- mkMult;
    Reg #(int) rg_x <- mkReg (1);
    Reg #(int) rg_y1 <- mkReg (1);
    Reg #(int) rg_y2 <- mkReg (1);
    rule gen_x;
    m1.put_x (rg_x);
    rg_x <= rg_x + 1; // Wertfolge
    endrule
    rule gen_y1;
    m1.put_y (rg_y1);
    rg_y1 <= rg_y1 + 2; // Wertfolge
    endrule
    rule gen_y2;
    m2.put_y (rg_y2);
    rg_y2 <= rg_y2 + 3; // Wertfolge
    endrule
    rule rl_connect;
        let x2 <- m1.get_w (); 
        m2.put_x (x2);
    endrule
        
    Reg #(int) rg_j <- mkReg (0);
    rule drain;
        let w2 <- m2.get_w ();
        $display ("Product [%0d]: %0d x %0d x %0d = %0d”,rg_j, rg_j+1, rg_j*2+1, rg_j*3+1, w2");
        if (rg_j == 10) $finish ();
        rg_j <= rg_j + 1;
    endrule

endmodule: mkTestbench 
endpackage

[image]

quark17, 2024-02-15:
Can you use the “attach files” option to upload the file? Because I’m not sure if the text is displaying correctly. For example, I notice an extra quote (“) at the end of the $display arguments?

quark17, 2024-02-15:
You also have not provided the source for the `Mult` package which is being imported.

However, I believe the error is because you are using a unicode character in a string somewhere, such as in a `$display` statement.  I can reproduce the error with this example:
```
(* synthesize *)
module mkTest ();
  rule r;
    $display("ă");
    $finish(0);
  endrule
endmodule
```

This is something that we overlooked when we recently added Unicode support.  We will look at it in the coming weeks, but in the meantime, characters with encodings above 255 should be avoided in strings.

The cause of the problem is [here](https://github.com/B-Lang-org/bsc/blob/1c27a41ea5d6fbcd0a27c207a117512df37a0c1c/src/comp/Util.hs#L128) (line 128 in `src/comp/Util.hs`).  This is used to define the function `to_quoted_string`, which is used in both `CCSyntax.hs` and `Verilog.hs` to print string literals in C++ and Verilog.  The code currently gives an internal error if the character encoding is above 255.  That should be fixed.  Though off hand I don't know if the answer is for `to_string` to print an encoding (like `\u1234`), or to print the unicode character directly, or to not use `to_string` and instead have a function that returns a byte encoding -- and possibly the C++ and Verilog contexts should use different functions, if the syntax for encoding of strings is different between the two.  (Mostly I think that `to_string` is trying to escape characters like `\n` and `\t` etc.)

The error message is expected to print the encoding number that was found, but it is not being printed because of missing parentheses in line 128 of `Util.hs`, around the string concat that adds the number to the error message.  Because the context is expecting a string, the expression still typechecks, so the missing parentheses isn't caught!