B-Lang-org/bsc #696: Failures with compilation on raspberry-pi 5

mycroft92, 2024-05-10:
When compiling on raspberry-5 with raspbian OS, I get the following errors.
```
ghc -Wtabs -fmax-pmcheck-models=800 -hidir /home/mycroft92/bsc/src/comp/../../build/comp -odir /home/mycroft92/bsc/src/comp/../../build/comp -stubdir /home/mycroft92/bsc/src/comp/../../build/comp -main-is Main_bsc \
	-O2 -hide-all-packages -fasm -Wall -fno-warn-orphans -fno-warn-name-shadowing -fno-warn-unused-matches -package base -package containers -package array -package mtl -package unix -package regex-compat -package bytestring -package directory -package process -package filepath -package time -package old-time -package old-locale -package split -package syb -package integer-gmp -package text  -iGHC/posix -iLibs -i../Parsec -i../vendor/stp/include_hs -i../vendor/yices/include_hs -i../vendor/htcl '-tmpdir /tmp'  -I../vendor/stp/include -I../vendor/yices/include -L../vendor/htcl -I/usr/include/tcl8.6  -lpthread --make bsc -j1 +RTS -M4G -A128m -RTS "-with-rtsopts=-H256m -K10m -i1" -rtsopts -L../vendor/stp/lib -lstp -L../vendor/yices/lib -lyices 

when making flags consistent: warning:
    Native code generator doesn't support target platform, so using LLVM
[136 of 224] Compiling Parser.BSV.CVParser ( Parser/BSV/CVParser.lhs, /home/mycroft92/bsc/src/comp/../../build/comp/Parser/BSV/CVParser.o )
^[[3~`opt-13' failed in phase `LLVM Optimiser'. (Exit code: -9)
make[2]: *** [Makefile:297: bsc] Error 1
make[2]: Leaving directory '/home/mycroft92/bsc/src/comp'
make[1]: *** [Makefile:61: install] Error 2
make[1]: Leaving directory '/home/mycroft92/bsc/src'
make: *** [GNUmakefile:41: install-src] Error 2
```

Here are the system details:
```
$ uname -a
Linux mycroftPi 6.6.28+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux
```
If the issue is optimisation flag related with `-O2` then is there a clean way I can disable optimisations and retry?

Thanks in advance.

quark17, 2024-05-15:
You can define `GHCOPTLEVEL` to an empty string (or to `-O0` in case the default is not zero).  This would override the definition in `bsc/src/comp/Makefile`:
```
GHCOPTLEVEL ?= -O2
```
However, the error message is for `opt` which is an LLVM tool and not GHC, so I don't know if the GHC optimization level will affect the use of `opt`.  A web search for that error message shows that it has been caused in the past by a segfault in `opt`, and I see a closed issue on GHC's repo, but the result was just to open a bug on LLVM's repo (which was later closed).

I'm surprised that you're running GHC on a raspberry pi.  Are you able to successfully run GHC to compile other programs besides BSC?  I would typically run BSC on a beefier system and then move the generated Verilog to where it's needed, but I'm also not familiar with raspberry pi 5.

mycroft92, 2024-05-15:
Thanks for the reply. 

Pi 5 is reasonably powerful to run GHC (with a 4GB RAM). I switched the OS to ubuntu server and the entire toolchain works (compilation included) without needing to change any customizations. You may please close this ticket.

I agree with your second comment (about moving the verilog files to pi) because the compile times are high on pi. My original intent was to have CI/CD flow on the raspberry pi with an FPGA connected to it. My current system (MacOS Sonoma) doesn't have support for Xilinx vivado.

quark17, 2024-05-16:
I'm fortunately running on an old x86 Mac, so I am able to run Altera tools in a VirtualBox VM.  But for ARM-based Mac, I'm not aware if an x86 emulator exists, and I'd guess it might be slow, if it does?

closed 2024-05-16 by quark17

mieszko, 2024-05-16:
I've used [UTM](https://mac.getutm.app/) (which wraps qemu) to run x86 Linux VMs on arm Macs (GUI-less). Don't expect stellar performance but I've run Modelsim and Quartus this way on some not-terribly-large designs.