B-Lang-org/bsc #725: how does bsc disambiguate between V2K keywords and library functions?

iamkarthikbk, 2024-07-30:
I am trying to detect the presence of an element in a vector using multiple O(N) comparators. I know that there are simpler ways of doing this, but why does the below example not compile?

```bsv
import Vector :: * ;

module mk_empty(Empty);
   Vector#(10, UInt#(4)) v_datastore = {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15};
   UInt#(4) lv_search_value = 5;
   function Bool fn_match(UInt#(4) search, UInt#(4) heap) = (search == heap);
   Bool its_there = or(zipWith(fn_match, replicate(lv_search_value), v_datastore));

   rule rl_print;
      $display("its_there: ", fshow(its_there));
      $finish;
   endrule
endmodule
```

i get the error message
```shell
Unexpected V2K keyword `or'; expected expression or `('
```

as far as i can see, this should work
[image]

quark17, 2024-07-30:
BSV has some keywords (including many from SystemVerilog) that BH does not, so some names that are available in BH are sometimes more difficult to use in BSV.  In the Prelude, there is a function `const` which conflicts with a SV keyword, so we have defined `constFn` as an alias which can be used easily in BSV.  We might consider adding `orFn` to the Vector and List libraries.

Note, though, that you can still use these names, but you have to use the SystemVerilog-style of escaping the name, using a backslash.  For example, this should work in your code:
```
Bool its_there = \or (zipWith(fn_match, replicate(lv_search_value), v_datastore));
```
The backslash indicates that all non-space characters that follow it are part of the identifier name -- including symbols like parentheses.  So make sure to put a space after the name, so that the arguments to the function aren't included in the name!

Until something like `orFn` is added to the library, you can also define it in your own files:
```
function Bool orFn (Vector#(vsize,Bool) vect);
  return \or (vect);
endfunction
```

iamkarthikbk, 2024-07-31:
ah, that makes sense. Thank you, quark17!