B-Lang-org/bsc #732: Rules with conflicts can happen at the same cycle

uv-xiao, 2024-08-28:
I encounter a situation where two rules that have conflicts (writing to the same register) can happen at the same cycle.  The design and testbench are as follows:
```
interface DelayInterface;
  method Action feedSignal(Bit#(4) signal);
  method ActionValue#(Bit#(4)) get;
endinterface

(* synthesize *)
module mkDelayModule (DelayInterface);
  Reg#(Bit#(4)) delay1 <- mkReg(0);
  Reg#(Bit#(4)) delay2 <- mkReg(0);
  rule often;
    delay2._write(delay1<<1);
    $display("often: enq %d*2", delay1);
  endrule
  rule odd ((delay1 &1) == 1 );
    delay2._write(delay1);
    $display("odd: enq %d", delay1);
  endrule
  method Action feedSignal(Bit#(4) signal);
    delay1 <= signal;
  endmethod
  method ActionValue#(Bit#(4)) get;
    return delay2;
  endmethod
endmodule

module mkTestbench(Empty);
  DelayInterface delay <- mkDelayModule;
  Reg #(int) step <- mkReg(0);
  rule rl_step;
    step <= step + 1;
    if (step == 6) $finish;
  endrule
  rule rl_1 (step <= 4);
    Bit #(4) signal = truncate(pack(step+1));
    delay.feedSignal(signal);
    $display("%2d: feed %2d", step, signal);
  endrule
  rule rl_get (step >= 2);
    let signal = delay.get();
    $display("%2d: get  %2d", step, signal);
  endrule
endmodule
```
The bsim give the following output:
```
often: enq  0*2
 0: feed  1
often: enq  1*2
odd: enq  1
 1: feed  2
 2: get   1
often: enq  2*2
 2: feed  3
 3: get   4
often: enq  3*2
odd: enq  3
 3: feed  4
 4: get   3
often: enq  4*2
 4: feed  5
 5: get   8
often: enq  5*2
odd: enq  5
 6: get   5
often: enq  5*2
odd: enq  5
```
It shows that both the rule `often` and `odd` happen in steps 1, 3, and 5, where the `odd` rule shadows the `often` rule. I wonder why this would happen.

I appreciate any help you can provide.

mieszko, 2024-08-29:
so this is how the Reg semantics are defined to be, isn't it?

from `src/Libraries/Base1/Prelude.bs` in lines 1846 et seq:

```
vMkReg v =
    module verilog "RegN" (("width",valueOf n), ("init",v)) "CLK" "RST" {
        read = "Q_OUT"{reg};
        write = "D_IN"{reg} "EN";
    } [ read <> read,
        read < write,
        write << write ]
```

basically what the last line says is that if you have two rules that write diff values to the same reg then bsc can pick one of them to logically happen "before" the other "in the same cycle," which means that the "logically later" write will trump the other one.

note this only happens when there are no ordering constraints in the other direction, though. if you had two rules where one copies reg x to reg y and the other copies y to x, they _won't_ get scheduled in the same cycle because there is no way to pick the one that happens "logically before."

basically the question that the bsc scheduler is asking is, is there a way to schedule some rules in one cycle so that the overall outcome can be explained by _some sequence_ of these rules being applied individually one after the other? if so, it will schedule them together. this is how the rule semantics work.

a trivial way for this to happen is of course if the rules don't access the same state. a less trivial (but still trivial) way is if both rules _read_ the same state; you can see this in `<>` above.

uv-xiao, 2024-08-30:
Thanks a lot for the clear explanation!

closed 2024-09-10 by uv-xiao