B-Lang-org/bsc #737: Add back middle-dot as function composition for backward compatibility

kenta2, 2024-09-11:
I've encountered legacy code that uses middle-dot as function composition (someone had created a forked version of some of the BSC standard libraries), and it would be nice to still be able to compile that without too much trouble.

This commit https://github.com/B-Lang-org/bsc/commit/369041ec1f347a4ecf44736e54598d3852f11151 changed all instances of middle-dot (U+B7) to ring-operator (U+2218).  But we didn't have to stop supporting middle-dot; we can support both middle-dot and ring-operator both for function composition.

(`iconv` can handle converting from ISO 8859-1 to UTF-8.)

krame505, 2024-09-11:
Can we support this without changing back to ISO 8859-1?  I recall prior to this change, editing Prelude.bs was rather annoying.  My editor insisted on changing the file encoding whenever I saved, which caused any appearance of U+B7 to get corrupted.

krame505, 2024-09-11:
See https://github.com/B-Lang-org/bsc/pull/601, for reference.

kenta2, 2024-09-11:
Yes, we can support this without changing back to ISO 8859-1.  The middle dot exists in unicode, and can be encoded in UTF-8.

quark17, 2024-09-17:
What about adding a flag to BSC to specify additional "prelude" libraries, that are automatically imported.  Then you could define the middle-dot compose operator in a file `MyPrelude.bs` and not have to edit any source files to add an explicit import of `MyPrelude` (or `Utils` if you prefer).

BSC already has a hidden flag `-use-prelude` that controls whether the prelude libraries are implicitly imported or not; the flag is on by default and is used as `-no-use-prelude` during the compile of BSC's prelude, to avoid a recursive import.  We could add a flag like `-prelude <packages>` and support the `+` symbol to indicate the default packages (like we do with paths to specify the default path), so a user could add a package with `+:MyPrelude` and could omit the prelude using an empty list.

Would that satisfy your need to support an old codebase without having to make manual edits to many files?