B-Lang-org/bsc #760: Failed to import existing RTL into a  BSV design

chenshih1, 2025-01-25:
I am reading the book "BSV by Example" and trying to run the example code provided in each chapter to learn the features of Bluespec SystemVerilog. The example code from Chapter 15, "Importing existing RTL into a BSV design," cannot be successfully compiled, and the errors encountered with FIFOifc and GetPutifc are not the same. So, how can one correctly import existing RTL into a BSV design?

The operating system I am using is Ubuntu 22.04.5 LTS, and the version of the bsc compiler is Bluespec Compiler, version 2024.07 (build b4f31db).

When I tried to compile FIFOifc and run Bluesim simulation, I encountered the following errors:
```
Linking for Bluesim ...
bsc -e mkTb -sim -o mkTb_bsim -simdir build_bsim -bdir build_bsim -info-dir build_bsim -keep-fires -p .:%/Libraries
build_bsim/mkTb.cxx: In member function ‘void MOD_mkTb::reset_RST_N(tUInt8)’:
build_bsim/mkTb.cxx:43:10: error: ‘class MOD_Fifo<unsigned char>’ has no member named ‘reset_V_RST_N’; did you mean ‘reset_RST’?
  43 |   INST_i.reset_V_RST_N(ARG_rst_in);
       |          ^~~~~~~~~~~~~
       |          reset_RST
make: *** [Makefile:60: link] Error 1
```

The Verilog simulation of FIFOifc encountered the following errors:
```
Linking for Verilog sim ...
bsc -e mkTb -verilog -o ./mkTb_vsim -vdir verilog_dir -bdir build_v -info-dir build_v -vsim iverilog verilog_dir/mkTb.v
verilog_dir/mkTb.v:42: error: parameter `V_GUARDED` not found in `main.top.i`.
verilog_dir/mkTb.v:39: error: parameter `V_P1WIDTH` not found in `main.top.i`.
verilog_dir/mkTb.v:40: error: parameter `V_P2DEPTH` not found in `main.top.i`.
verilog_dir/mkTb.v:41: error: parameter `V_P3CNTR_WIDTH` not found in `main.top.i`.
5 error(s) during elaboration.
ERROR: cannot compile Verilog files
make: *** [Makefile:92: v_link] Error 5
```

The errors encountered during the Bluesim simulation and Verilog simulation of GetPutifc are shown in the following two logs respectively:
```
Linking for Bluesim ...
bsc -e mkTb -sim -o mkTb_bsim -simdir build_bsim -bdir build_bsim -info-dir build_bsim -keep-fires -p .:%/Libraries
Internal Bluespec Compiler Error:
Please report this failure to the BSC developers, by opening a ticket
in the issue database: https://github.com/B-Lang-org/bsc/issues
The following internal error message should be included in your
correspondence along with any other relevant details:
Bluespec Compiler, version 2024.07 (build b4f31db)
Fifo expected module SizedFIFO to have FIFO width and depth arguments but it was given arguments:  [32'd8,
                                                                                                  32'd2,
                                                                                                  32'd2,
                                                                                                  1'd1,
                                                                                                  32'd0]
make: *** [Makefile:60: link] Error 1
```
```
Linking for Verilog sim ...
bsc -e mkTb -verilog -o ./mkTb_vsim -vdir verilog_dir -bdir build_v -info-dir build_v -vsim iverilog verilog_dir/mkTb.v
verilog_dir/mkTb.v:42: error: parameter `V_GUARDED` not found in `main.top.i`.
verilog_dir/mkTb.v:39: error: parameter `V_P1WIDTH` not found in `main.top.i`.
verilog_dir/mkTb.v:40: error: parameter `V_P2DEPTH` not found in `main.top.i`.
verilog_dir/mkTb.v:41: error: parameter `V_P3CNTR_WIDTH` not found in `main.top.i`.
5 error(s) during elaboration.
ERROR: cannot compile Verilog files
make: *** [Makefile:92: v_link] Error 5
```

The Makefile I am using is as follows, modified from [https://github.com/bsvlang/main](url):
```makefile
# ================================================================
# Please modify the following for your installation and setup

# Set this to the command that invokes your Verilog simulator
VSIM ?= iverilog
# VSIM ?= verilator

DISTRO         ?= .
TOPFILE        ?= $(DISTRO)/Tb.bsv
TOPMODULE      ?= mkTb
[... 196 lines cut ...]
    input_reset rst_RST_N (V_RST_N) clocked_by (clk) <- exposeCurrentReset;

    interface Get g;
        method V_D_OUT get () enable (V_DEQ) ready (V_EMPTY_N);
    endinterface

    interface Put p;
        method put (V_D_IN) enable (V_ENQ) ready (V_FULL_N);
    endinterface

    schedule g_get CF p_put;
    schedule g_get C g_get;
    schedule p_put C p_put;

endmodule: mkSizedFIFOtest
           
(* synthesize *)
module mkTb (Empty);

   MyGetPut #(Bit #(8)) i <- mkSizedFIFOtest (2, True);

endmodule: mkTb

endpackage: Tb
```

quark17, 2025-01-26:
It is our intention to put the source files for _BSV By Example_ (both the document and the examples) into this repository.  When we do that, the examples will be set up with Makefiles and test benches, and verified as part of the BSC regression testing.  Unfortunately, we have not had time to do that yet, so I apologize that you ran into problems.

The problems are because `SizedFIFO` is the name of a module in BSC's standard libraries.  The ports and parameters on BSC's module have different names from the `SizedFIFO` being imported in the example.  So if BSC accidentally uses the module from the library, you get errors due to the names not matching in the instantiation.

I suggest that you renamed the `SizedFIFO` Verilog module to something else, such as `MyModule` (or `MySizedFIFO`).  If you do that, then you'll get more sensible error messages.

(1) For Verilog linking, the BSC command you are using doesn't provide the location of the imported module.  If you change the name to `MyModule`, you'll see that you get an error that the module is not found:
```
verilog_dir/mkTb.v:42: error: Unknown module type: MyModule
```
BSC has a path of directories that it searches to find Verilog modules.  The directory with BSC primitives (`inst/lib/Verilog/`) is in that list -- so if BSC doesn't find `SizedFIFO.v` in the current directory, then BSC will find the file by the same name in the BSC library.  Normally, the current directory is first in the search path, and so the local `SizedFIFO.v` file would be found first.  However, you have provided the `-vdir` flag, which causes BSC to put the vdir directory first, in place of the current directory (which is no longer in the path).  If you want `SizedFIFO.v` (or renamed as `MyModule.v`) to be found by BSC, while also providing the `-vdir` flag, then you either need to provide the file on the command line or add its directory to BSC's search path, as follows:
```
// Specify a Verilog file on the command line
bsc -verilog -e mkTb ... verilog_dir/mkTb.v ./SizedFIFO.v

// Specify a Verilog search directory on the command line
bsc -vsearch .:+ -verilog -e mkTb ... verilog_dir/mkTb.v
```
The `-vsearch` flag takes a list of directories separated by a colon (`:`).  It will replace the current search path with the new value, so if you want to add to an existing path, then use `+` to refer to the previous value.  In the example above, `.:+` is adding the current directory (`.`) to the front of the existing search path.

With one of those fixes, the Verilog linking should be fixed.

(2) Bluesim does not support importing Verilog, so that is not expected to work.

Normally, BSC will give a more helpful error, like the following:
```
Error: "Tb.bsv", line 39, column 22: (G0084)
  Bluesim does not currently support imported Verilog modules. The
  instantiation `i' in parent module `mkTb' is an imported Verilog module
  `MyModule'.
```

However, there are Bluesim implementations for BSC's primitive modules.  So BSC does know about `SizedFIFO` and allows an import-BVI statement for it, because that's how the primitive is declared in BSC's libraries.  Unfortunately, BSC doesn't currently distinguish between those imports in the BSC libraries and imports in the user's code.  So when you try to import a module called `SizedFIFO`, while compiling for Bluesim, BSC allows it, and assumes that you are using the primitive Bluesim model.  That model has different port and parameter names, which is why you get the error.

This is an issue that we should fix -- we should make it so that BSC's primitive module names don't clash with user module names.  Sorry that you ran into this issue.  If the example had been constructed with different module name and a pre-written Makefile, that would also have avoided the issues.

chenshih1, 2025-01-27:
@quark17 After following your advice to rename the Verilog files and adjust the Makefile, the Bluesim simulation now presents clearer error messages, while the Verilog simulation runs smoothly. I greatly appreciate your assistance.

quark17, 2025-02-07:
I'm glad that helped!

I will close this issue, because I found that Issue #139 is already open for the problem of name clash between user imported modules and library primitives.  BSC could be changed to distinguish between import-BVI statements in the library and import statements by the user, but that would only solve the name clash within BSC; the generated Verilog would still have a flat name-space and the module names would clash later in the flow when using Verilog tools.  Renaming the Verilog modules solves both of those clashes.  The proposal in #139 is to rename the library modules with a prefix like `__BSC_`, so `SizedFIFO` becomes `__BSC_SizedFIFO`.

closed 2025-02-07 by quark17