B-Lang-org/bsc #769: Bluespec evaluation-time error: unhandled case

jahagirdar, 2025-03-26:
With Bluespec Compiler, version 2024.07 (build b4f31dbe) the attached file generates the following error

```
bsc -elab  -verilog  I2C.bsv

Compilation message: "StmtFSM.bs", line 220, column 72: Case: [SNamed sendbytes_l102c63 [[Par_l102c63 ([Action_l103c54] [Action_l104c57] [Seq_l105c49 ([While_l106c57 [SNamed sendbits_l106c76 [[Par_l106c76 ([Action_l107c73] [SExprS _l108c77])]]]] [SExprS _l110c65])])]]]
Error: "StmtFSM.bs", line 964, column 13: (S0015)
  Bluespec evaluation-time error: unhandled case
  During elaboration of `c0' at "StmtFSM.bs", line 942, column 7.
  During elaboration of `c1' at "StmtFSM.bs", line 943, column 7.
  During elaboration of `c_no_action' at "StmtFSM.bs", line 642, column 8.
  During elaboration of `_rfsm' at "StmtFSM.bs", line 1140, column 8.
  During elaboration of `fsm' at "I2C.bsv", line 134, column 21.
  During elaboration of `mkI2CMaster' at "I2C.bsv", line 26, column 8.
```

[code.zip](https://github.com/user-attachments/files/19471137/code.zip)

jahagirdar, 2025-04-02:
Update on this issue:
The error is due to the labels on the seq/par blocks. Looks like the tool is not able to handle named blocks.

quark17, 2025-04-02:
Yes, that's my diagnosis.  Sorry that I didn't explain that in the email thread.  It's possible that this can be fixed by adding case arms to handle named blocks around line 961 in `StmtFSM.bs`:
```
getNoActionCondition (SNamed _ _ Nil) = return True
getNoActionCondition (SNamed _ _ (Cons st Nil)) = getNoActionCondition st
```
Are you able to try that out and see? and maybe submit a PR?