B-Lang-org/bsc #793: Simulation displays `Vector` of bytes in reverse order when using `%0s` modifier

RyanGlScott, 2025-07-03:
In the following program:

```bluespec
package Top where

import Vector

abcd1 :: String
abcd1 = "abcd"

abcd2 :: Vector 4 (UInt 8)
abcd2 = update (update (update (update newVector 0 97) 1 98) 2 99) 3 100 -- [97,98,99,100]

mkTop :: Module Empty
mkTop =
  module
    rules
      "r": when True ==> action
        $display " %0s" abcd1
        $display " %0s" abcd2
        $display (fshow abcd2)
```

I would expect each `$display` line to print something akin to `abcd`. Surprisingly, only two of the three lines actually do this:

```
$ bsc -sim -g mkTop -u Top.bs
$ bsc -sim -e mkTop -o mkTop.exe
$ ./mkTop.exe -m 1
 abcd
 dcba
<V  97  98  99 100  >
```

The first line prints `abcd`, as expected. The third line prints the raw `Vector` elements `97`, then `98`, then `99`, then `100`, which is also more-or-less what I would have expected.

The second line is the strange one. It prints `dcba`, which is the opposite order of what I would have expected! As the third line demonstrates, the actual `Vector` elements appear to be in the correct order—it's just that `$display` is showing them in the opposite order. Is this behavior intended, or is this a bug?

RyanGlScott, 2025-07-05:
Thanks for the explanations! I'm willing to accept that I'm misusing the `%s` modifier a bit here (and the explanation for how the elements end up in the reverse order makes sense.)

To complete the exercise I originally set out to do: is there a convenient way to convert a `Vector` of `UInt 8`s to a `String` in Bluespec so that I can pass the resulting `String` to `$display`?

quark17, 2025-07-05:
It depends on whether you're building strings at compile time or at run time of the hardware.  If you're building them at compile time, then you could use the type `String` (or possibly `Char`).  See the library `Base3-Misc/Printf.bsv` for an example of compile-time construction of strings -- including support for things like `%f` for `Real` values (related to your other question).

To build strings at run time, it depends on what you're doing.  The `Fmt` type (and `$format` and the `+` operator for concatenating `Fmt` values) exists precisely to allow building strings for display, that BSC can transform into an efficient sequence of conditional `$write` calls (rather than leaving as a large if-then-else argument into a single `$display`).  So I would suggest looking the various `FShow` instances to see how `$format` is used.

If you do want to generate hardware that passes strings around as bit-vectors (even storing them in registers), then using Verilog's `%s` format specifier is perfectly fine -- I wouldn't say it's a misuse.  You just need to adhere to Verilog's encoding format for strings, which starts reading the characters from the high end (MSB).  If you're using `Vector#(N,Bit#(8))`, then start filling in the characters at element `N-1`; alternatively, you can start at zero, as long as you call `reverse` on the vector before passing it to `$display`.  However, unless you have a need to store strings in registers, I wouldn't think you need to go down this route.  Using `String` and/or `Fmt` ought to work.